dresden-elektronik/deconz-rest-plugin #2542: Heiman HS1SA Smoke Detector

og9gjz, 2020-03-06:
Hi, 
i've bought 3 **Heiman HS1SA** and paired them with Phoscon App. The senors doesn't show up in the  Phoscon App but in deCONZ.
I'm using a Caonbee II with **264A0700** and Phoscon **App 2.05.74 / 25.2.2020**

In deCONZ the Heiman HS1SA have the Manufacturer Code "0x120b" the IEEEs are:
0x00158d0004c0a815 / 0x00158d0004c07f9b and 0x00158d0004c0a695
I hope you can add this sensor to the list of supported devices!

Is there a way to pair them to a group, so if one sensor detects smoke, all sensors will beep?

Sjonnie2018, 2020-03-06:
Have a look at issue #2373 and check the exact product description/number to see if that issue applies to your situation.

HTH

Greetz,
Sjonnie

Mimiix, 2020-06-05:
Duplicate issue with #2373. Closed for now.

closed 2020-06-05 by Mimiix