dresden-elektronik/deconz-rest-plugin #2544: IKEA Trådfri 250lm E27 Bulbs Incorrectly identified

scottsweb, 2020-03-06:


Smanar, 2020-03-07:
I think you have same problem than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2456

Was not your device that are detected but the conbee.

joch, 2020-03-09:
I have that light, and it's detected successfully, so it's probably related to the issue mentioned above.

scottsweb, 2020-03-10:
Thanks for the pointers. Just upgraded to `2.05.75` and the phantom device/Conbee is no longer appearing which is good (and less confusing) during a scan. 

I am still struggling getting these bulbs to pair though. Will keep trying! They pair with the remote just fine :/

scottsweb, 2020-03-10:
Changing the Zigbee channel to `11` seems to have fixed it. I wasn't able to change the Zigbee channel until this version `2.05.75` so that is progress to. Closing.

closed 2020-03-10 by scottsweb