dresden-elektronik/deconz-rest-plugin #2545: Several errors after installing Deconz and Hue-emulator

NightFalcon650, 2020-03-07:
Goodday,

After installing deconz and hue-emulator according the manual on github everything worked.
After rebooting i have to start the services by hand and they stil keep giving errors.

[image]

Whats wrong and what could i do to fix it?

SwoopX, 2020-03-07:
To me it looks like you are trying to start 2 deconz services simultaneously. "deconz.service" is for headless setups and "deconz-gui.service" for GUI mode, requiring and X-Server. Choose the one you need and do `sudo systemctl enable SERVICE`, for the other one `sudo systemctl disable SERVICE`. Afterwards, restart the serivce of your choice.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix