dresden-elektronik/deconz-rest-plugin #2546: Connections lost - Failing to update from 0x260b0500

andersbillesoebeck, 2020-03-07:
Hi All,

I am using Deconz with Hassio on a Rpi3 and ConBee. I have done so the last year or a bit more. It have always been rock-stable with about 20 lights (mostly Hue but a few IKEA) and about 10 Aquara sensors. For quite a while, updating ConBee have failed from Phoscon UI, so perhaps ConBee have never been updated.

After updating Hass to 105 and OS 3.09 and Deconz to 1.05.71 I lost connections to all Hue lights. Everything else worked fine. I tried re-creating earlier network configs, I bought a Dimmer Remote and reset the lights, nothing worked. 

In the logs, it seemed like Deconz tried to switch to channel 25 but failing, and shows an error that that ConBee should be updated., 

I completely removed the Docker, re-installed, switched to channel 25 on an empty setup and managed to recreate the network. But, after a reboot all connections are lost again and loggin-output shows “Channel is 15 but should be 25” errors.

Updating manually on my windows PC shows boot loader timeouts like in this
`action: update firmware after 2 ms
serial port open device
STARTING APP
(395 ms)
STARTING APP
?�� (475 ms)
STARTING APP
?�� (2605 ms)
could not sync with bootloader `

I hope an update can make my ConBee work with the newest Deconz again, as I assume this is the problem. Is there any way at all to make it update? 

I am getting desperate to make my light-system work again...

SwoopX, 2020-03-07:
Check out #2493

andersbillesoebeck, 2020-03-07:
I just hoped it was that simple.

I ran the suggested -t 60 to GCFlasher, using both version 3_10 and 3_05. 

Unfortunately it just kept looping and timing out:

`Reboot device COM3 (ConBee)
deCONZ firmware version 260B0500
reset FTDI DM01BAWAA (dev: 0)
action: update firmware after 4 ms
STARTIN (391 ms)
STARTING APP
 (408 ms)
STARTING APP
? (487 ms)
STARTING APP
? (2600 ms)
could not sync with bootloader
Reboot device COM3 (ConBee)
deCONZ firmware version 260B0500
reset FTDI DM01BAWAA (dev: 0)
action: update firmware after 3 ms
STARTIN (390 ms)
STARTING APP
 (406 ms)
STARTING APP
? (486 ms)
STARTING APP
? (2599 ms)
could not sync with bootloader
Reboot device COM3 (ConBee)
deCONZ firmware version 260B0500
reset FTDI DM01BAWAA (dev: 0)
action: update firmware after 3 ms
STARTING (390 ms)
STARTING APP
 (406 ms)
STARTING APP
? (486 ms)
STARTING APP
? (2599 ms)
could not sync with bootloader
retry, failed`

Any other good ideas?

SwoopX, 2020-03-07:
Gee, the firmware appears to be from mid 2016 or so... Eventually, you need an equally old GCCFlasher or maybe sequential firmware updates.

@manup Maybe you can help out here?

andersbillesoebeck, 2020-03-07:
Yes, I know. 

As I uset the old Hassio plugin, Updates did not work without a lot of hassle, which I never did. And as it were running in a live system I did not update on my PC (and it just worked fine and very robustly). Until just recently...

I am fine with a sequential update if that can fix my problem.

Any pointers from @manup will be appreciated to guide the approach.

SwoopX, 2020-03-08:
Which firmware are you trying to flash? Maybe you should start with the one following the one you're running on. See http://deconz.dresden-elektronik.de/deconz-firmware/ for all the files. Eventually, also an older version of GCCFlasher might be required, which you also can find there.

andersbillesoebeck, 2020-03-08:
I finally managed to update, based on your insight on using an older flasher. Using GCFlasher 2_10 I managed to flash first the version just after the one I had on. Then I tried 3_10 to flash the newest - it failed. Then I tried 2_10 to flash the newest - it worked. Then I tried using 3_10 to flash the newest again - then it worked!

So having the newest firmware, I then started from a completely new Deconz Docker. It recognized the newest firmware, but reported the interface as Raspbee. 

I started adding devices.. IKEA bulbs went without problems, but my problem were back: 

No Hue bulbs are detected... I reset them using the Dimmer Switch, but they remain silent. 

Any ideas?

I use default channel (15) which I also used when it worked. I think I used cannel 25 last time the Hues were reset and connected, when trying to recover my network.

SwoopX, 2020-03-08:
Awesome it worked out.

I'm afraid I cannot be too much of a help regarding your actual lights problem. Prefer to make my existing lights in general smart for energy efficiency reasons. Regardless, since I don't use any bulbs nor use docker, I'm lacking experience on that field. Have you already tried searching the issues? Btw, afaik, default channel is 25. Deconz version .75 is also out since today.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix