dresden-elektronik/deconz-rest-plugin #2549: I am using docker and the bind drop down box is grayed out

redwoodimage, 2020-03-08:
When I vnc into the container the bind drop down box is grayed out. Am I missing something what do I have to do to enable it.

redwoodimage, 2020-03-08:
I am using phoscon 2.05.74 and firmware 26490700, yes the conbee is detected. 

root@debian:/# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II 
[image]

Smanar, 2020-03-08:
Haaa ok^^ I understand.

You want to make a bind ?

Just use drag and drop. Wait 5mn I m searching the procedure.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix