dresden-elektronik/deconz-rest-plugin #2550: Aqara temperature sensor shows online even when battery removed

tuomassa, 2020-03-08:
So I was trying to debug why my sensors shows ~25 C all the time. I even removed the battery and it is still shown online?! How is it possible? How can I even trust the readings if it cannot even be recognized when there is no connection?

ebaauw, 2020-03-08:
The are no connections in ZigBee, only messages.  Technically, ZigBee end-devices are unreachable almost all the time, as they turn off their radio to preserve battery power.  They are only reachable for a fraction of a second when they wake up.  Light sleepers wake up every ~5 seconds (appearing to be reachable all the time, as their parent router caches any message for them for 7 seconds); deep sleepers might only wake up once an hour.

`state.reachable` or `config.reachable` is set when the gateway sees a message from the device, and cleared when the gateway hasn't seen any message from the device for a prolonged period of time.  This period depends on how frequently the gateway expects messages from the device.  For lights this should takes a couple of minutes; for deep sleepers, this might take several hours.

Check `state.lastupdated` on the `/sensors` resource to see when the gateway received the last message from the device.

ebaauw, 2020-03-15:
See #2590.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix