dresden-elektronik/deconz-rest-plugin #2552: Everything has stopped working (Raspbee)

redlegoman, 2020-03-08:
Around about midnight last night, my RASPBEE and Phoscon / deCONZ just stopped working, I couldn't turn off or on any lights and all my sensors stopped. All the devices are greyed out in the Phoscon app and everything in the deCONZ app is either yellow or grey. See screenshot. Everything was working just fine up until some point last night. In the phoscon web interface, I cannot change the channel and I cannot create a backup (it just says "Creating backup failed"). 
I have tried powering off the raspberry pi, unplugging the raspbee, turning off as many zigbee devices in  my home as possible, all to no avail. 

What can I do next?

[image]

[image]

[image]

[image]

[image]

[image]

In the last shot, even though some devices are not greyed out, opening or closing the sensors does not actually update in the interface. Everything is dead.

redlegoman, 2020-03-08:
I just tried the ALT-CLICK advanced options thing and loaded the config at the top of the list. I am now able to take a backup, but all the devices are still greyed out.

redlegoman, 2020-03-08:
If I scan for new lights I get a weird light show up: 
[image]

Not sure what that is. It wasn't there before.

Smanar, 2020-03-08:
The unknow 15 is normal, it's from the last version, it's the Raspbee.

manup, 2020-03-08:
Hi can you please try to restore the former configuration with the Zigbee configuration page.
The process is described in https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

redlegoman, 2020-03-08:
> 
> 
> Hi can you please try to restore the former configuration with the Zigbee configuration page.
> The process is described in https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

I did this and things came back, but I had to go around and reset every single device in my house, and some of the switches got different names, which messed up some automations in Home Assistant which was a major pain in the a*se.

redlegoman, 2020-03-08:
> 
> 
> The unknow 15 is normal, it's from the last version, it's the Raspbee.

When you mean the last version, does that mean the last software version?

Smanar, 2020-03-08:
This one > https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_74

And it's strange, I don't thing HA use deconz name but UniqueID, so you can chande device name in deconz without breaking HA. But I can't be wrong, I don't use HA.

redlegoman, 2020-03-08:
I mean why is there this unknown device? What is it exactly. It's not a problem, I'm just curious. You said "it's from the last version", I don't know what you mean.

Smanar, 2020-03-08:
> Support RaspBee, ConBee, ConBee II, XBee under lights resources 8c8e4b

SAOPP, 2020-03-11:
I can confirm one thing, the last two weeks I saw strange behaviours of my contact xiaomi sensor, when I open the door, but sensor still closed... after I open it twice it get actions...

I dunno what happened, 'coz I have only updates of my addon is come... and for half year afl worked like a clock. So now I have at 50 nodes, and start to repairing some nodes to my conbee 2, who was idle idle to somehow unload an existing network.

SAOPP, 2020-03-11:
Also today morning I faced with similar situation as of author os topic. Switches, door sensors, motion sensors simply did not respond, I had to restart the add-on Deconz... sheeeet.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix