dresden-elektronik/deconz-rest-plugin #2555: Friends of Hue switch does not connect to routers

philippsandhaus, 2020-03-08:
I am experiencing a problem with a Gira Friends of Hue switch. I had a lot of problems trying to pair and realized that it was only possible when placing it close to the coordinator (Conbee 2 stick). It also only operates reliable, when close to the stick although there a many more router devices (ikea light bulbs) which are far more close to the switch. I therefore suspect, that the switch does not have a link to these lightbulbs. Various other sensor and switch work fine also without direct connection to the Conbee stick.

The reason why I can only guess is that I cannot see the switch in the deconz guy via VNC. The switch is however visible and can configured in phoscon.

Can someone give me a hint? In my current setup the conbee stick and the switch are on the opposite ends of my home and the connection is very unreliable at the moment. I would really like to be able to use the mesh network and let the switch be connected to a router close to it.

ebaauw, 2020-03-08:
ZGP devices don't show up in the GUI.  deCONZ doesn't (yet?) configure the ZGP capable routers, so ZGP devices need to be in direct reach of the coordinator.  See #436.

philippsandhaus, 2020-03-09:
for the link and the explanation. I was not aware of this limitation for ZGP devices.

closed 2020-03-09 by philippsandhaus