dresden-elektronik/deconz-rest-plugin #2557: Hue Deep Dimming support for Hue Lights

00lex, 2020-03-09:
Hi! Hue released a new firmware for the bridge with improved deep dimming which means dimming <5%. 
https://www2.meethue.com/de-de/support/veroffentlichungshinweis/bridge
its a bridge thing. last lights firmware is from 2018. 

any chance to get this with deconz?

ebaauw, 2020-03-09:
What’s the observable difference with before?

00lex, 2020-03-09:
https://www.reddit.com/r/Hue/comments/fcsnno/comment/fjg941x

ebaauw, 2020-03-09:
> its a bridge thing.

I don't think so.  As far as I can tell from the sniffer, the Hue bridge still sends the same _Move to Level_ command to the bulb.  I don't think there's anything deCONZ can or should do differently.

> any chance to get this with deconz?

Update the light firmware (through the Hue bridge).

I reconnected one of my LCT003 spots to the Hue bridge to update its firmware to 5.130.1.30000.  I see no difference in minimum brightness between the updated spot and the others, running on 5.127.1.26420.  
From the firmware release notes, the deep dimming improvements only apply to a number of fixtures,  not to other bulbs nor spots.  The reddit post mentions one of those fixtures, a Being.

00lex, 2020-03-09:
whoops. thank you for clarification

closed 2020-03-09 by 00lex