dresden-elektronik/deconz-rest-plugin #2558: IKEA-shutters gone after update to 2.05.75

Michaelnorge, 2020-03-09:
After update to v2.05.75 all my shutters were gone from the Phoscon-App.
It was not possible to reconnect them, they didnt shown up again.

I made a downgrade to v2.05.74 - now everything works fine again.

dwarf-rbi, 2020-03-09:
I have the same issue.. all blinds are gone, but the REMOTE open/close are now showing..

manup, 2020-03-09:
Ah sorry our bad, since we're working on a dedicated window blinds page they are filtered on lights section (but are still available in the REST-API). Will be visible in the next version again until the new page officially launches.

dwarf-rbi, 2020-03-09:
@manup Thanks for the answer. Are there already a timeframe for when the next version is coming? Got 7 more blinds to add soon.

manup, 2020-03-09:
Should be online within the next 3 days, with only minor changes.

ebaauw, 2020-03-09:
> since we're working on a dedicated window blinds page they are filtered on lights section

Are you using the right filter?  The `lumi.curtain` is still visible in Phoscon.  This device also has a `type` of `"Window covering device"`.  I think some-one added `"Window covering controller"` as well.

Kane610, 2020-03-09:
@manup will that also bring proper pairing with the blinds to set up all bindings?

dwarf-rbi, 2020-03-10:
And maybe even fixing the current cover position bug. All of my blinds are at 100% all the time.

Kane610, 2020-03-10:
> And maybe even fixing the current cover position bug. All of my blinds are at 100% all the time.

That's what the binding does

wettermann32, 2020-03-12:
same problem, ikea shutters are gone. anyhow there was also the problems with only a down arrow to controll the shutters in homeassistant. Does this fix belong to this also?

dwarf-rbi, 2020-03-12:
@wettermann32 Im pretty sure the gray arrow is there because HA dosnt know the current position, and therefore thinks its all ways closed. So yes, this is does belong to this fix

dwarf-rbi, 2020-03-15:
@manup Any news on the new update? My 11 blinds dosnt really work at the moment :-(

llonca13, 2020-03-16:
Impossible to pair Fyrtur for days with this new version. After downgrade to v2.05.74, pairs and works again.

StoricU, 2020-03-18:
Any news about next update?
Considering downgrading to .74

wettermann32, 2020-03-22:
@llonca13
can you short describe how I can downgrade to 74?
Thanks Michael

Michaelnorge, 2020-03-23:
> @llonca13
> can you short describe how I can downgrade to 74?
> Thanks Michael

Turn on "expert-mode" at the adapter site.
Press the invert "+" on the far right in the line where the adapter is.
Choose the right version.

Zapfmeister, 2020-03-24:
Im also looking forward to the next release! :-)

dwarf-rbi, 2020-03-24:
@Michaelnorge Is this in the deCONZ ui you are referering to?

yabbah, 2020-03-27:
"Turn on "expert-mode" at the adapter site.", where do I found this?

wettermann32, 2020-04-12:
@manup: any news about the next version?

johnnykarlberg, 2020-05-01:
can someone please tell me how to downgrade in hassio without a snapshot?

frenck, 2020-05-01:
@johnnykarlberg You can't downgrade it without a snapshot.

johnnykarlberg, 2020-05-01:
Not even install a Fresh one with folder?

stigvi, 2020-05-04:
> Should be online within the next 3 days, with only minor changes.

What happened?

dwarfadmin, 2020-05-04:
Im not sure what happened @manup, Im still struggling with this current version

stigvi, 2020-05-04:
Well, in these Corona times, you never know. I hope everything is well with manup and everyone else who is contributing and their families.

monster400, 2020-05-09:
Hello

Any news? about next release?

llonca13, 2020-05-09:
+1. Could you at least say whats the status of some of the Ikea issues with deconz and if your not able to support it further to look elsewhere.. I dont want to sound rude, but this is not some no-name off brand chinese product, I bet half of conbee/raspbee buyers have this blinds and it would be very nice to update us on the status of this problem, or give us any info on the timeframe when this issue can be looked at. Thanks!

Michaelnorge, 2020-05-10:
I´am still on 74. 
I´am using Ikea shutters through the Ikea gateway and adapter and I´am happy with it so far.

yabbah, 2020-05-10:
Im on 2.05.75 and my IKEA Fyrtur working for me. But Phoscon dont have support for the open/close switch in the switch editor. But you can just use automations in HA och Node-red to get it working. I used these steps:

1. create the group for the blind in Phoscon
2. Start search for a light
3. Remove the battery in the blind for 5 seconds
4. Hold both bottons on the blind until it stop pulse, and 3 more seconds after that. 
After some seconds you will see should see the blind in Phoscon and deCONZ gui.
Change to a proper name and assign to the created group in step 1.

To add the open/close you just search for a switch -> Other and press the switch reset button 4 times within some seconds.

ebaauw, 2020-05-10:
The open/close remote can control the blinds directly, without the need for any automation.  Just add the blinds to the group associated with the remote.

yabbah, 2020-05-10:
> The open/close remote can control the blinds directly, without the need for any automation. Just add the blinds to the group associated with the remote.

I wasnt able to add the switch to the group in Phoscon tho.

Smanar, 2020-05-10:
@yabbah You don't need to add the switch to the group, but add the blinds to the group associated to the remote.

dwarf-rbi, 2020-05-10:
@yabbah You can add the open/close in phoscon. I wasnt able to connect the open/close directly to the blind. 
[image]

wettermann32, 2020-05-15:
deconz is not reporting the Position of the blinds. So I can only move it once down in hassio via the UI. 

It is now 2 months since manup said "within 3 days"… I hoped on more support using deconz Hardware. Has somebody experience with cc2531 and zigbee2mqtt?

hwikene, 2020-05-15:
Try removing batteries for 2-3min and seeing that helps

PexJensen, 2020-05-24:
@manup is this fixed in .76 ?

stigvi, 2020-05-24:
They are listed as light in .76

wettermann32, 2020-06-01:
Position of blinds is still not reported in .76. ALways 100%. You can Change the Position by Hand in hass via developertools-> Status. But after Restart you need to do this again.

Mimiix, 2020-06-12:
@wettermann32 You need to redo the binding for this to report. Info on that is in discord, or wiki page.

I'll close this issue, as its resolved.

closed 2020-06-12 by Mimiix

maqf12, 2020-06-20:
Turn on "expert-mode" at the adapter site.
Press the invert "+" on the far right in the line where the adapter is.
Choose the right version.

... Can someone pls tell me where do i find this ?.. What is adapter site ? ,, is it "gateway" where i can se firmware etc.. if so, there is no "+"