dresden-elektronik/deconz-rest-plugin #2560: Hue: Alexa App or Google Home App support?

NePheus, 2020-03-09:
I just bought a conbee 2 stick und installed my hue lamp. Now I tried to add the Hue Bridge inside the Alexa or Google Home App on Android. One of them redirects me to meethue.com that das not recognize my Hue Bridge and the other tells me to install the ofiicial Hue App, that finds my bridge, but dont let me continue, because it wants a Hue Bridge update (thats unfortunately not working when I click 'update'). Also when I try to add a "other device" in Alexa and turn the '3rd party apps' for 60 seconds on in the Phoscon app, it does not find anything. Are these apps not supported or am I doing something wrong?

NePheus, 2020-03-14:
**Background**
My thoughts in buying this stick was to extend the features of one local smart home server, which is my raspberrypi 3 running raspbian. The first step was the RPI-RF-MOD (Homematic CCU), the second the conbee 2 stick. Combining all this in one software, in my case FHEM.

**Topic**
So I wanted to find a way to use voice assist and the first idea was the official apps, that was unfortunately not working with my setup. I just asked myself why I am not using some FHEM feature of voice assist and came across "FHEM Connect". The Google Home app has also a specific plugin for this, also called "FHEM Connect". Now this ticket is solved for me.

closed 2020-03-14 by NePheus

lambu76, 2020-10-26:
Hi, could you please let me know which specific plug in you installed starting from FHEM connect ?

thx

NePheus, 2020-10-27:
I used this guide:
https://wiki.fhem.de/wiki/Google_Assistant_FHEM_Connect
It automatically recognizes the hue devices and displays it in the google home app.