dresden-elektronik/deconz-rest-plugin #2562: Still seeing old firmware in log, how to update?

pergolafabio, 2020-03-09:
Hi,

i am using this as add-on in HA, when i start the add-on, i still see this below in log : 

```
18:38:14:689 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
18:38:14:689 GW firmware version: 0x26330500
18:38:14:689 GW firmware version is up to date: 0x26330500
```

allthough updated already to .75  it should say, that firmware 0x263**50**500 should be available
so i am also unable to use the "update" http, since there is none :)

so how to update in docker?

thnx

manup, 2020-03-09:
Since this is a test firmware it isn't enabled by the update button yet.
You need to install it manually:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

manup, 2020-03-09:
Or in Docker container:

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

pergolafabio, 2020-03-09:
yeah, i was viewing that readme, but i dont see the new firmware in the logs as described here : 

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

i still see this in my docker ....

18:38:14:689 GW firmware version: 0x26330500
18:38:14:689 GW firmware version is up to date: 0x26330500

manup, 2020-03-09:
The logs only list the hard coded stable firmware which is currently 0x26330500.
But the file is there, it should be possible to use it with the command in step 3)

pergolafabio, 2020-03-09:
ahh ok , didnt know that

so 40500 , the one before was also a "test" version ? maybe i should wait then untill next stable if i dont have issues?

manup, 2020-03-09:
The firmware is only interesting if you facing issues :) So never change a running system.

pergolafabio, 2020-03-09:
ok thnx for the advice :)

closed 2020-03-09 by pergolafabio