dresden-elektronik/deconz-rest-plugin #2563: Upgrade to 2.05.75 breaks OpenHab Communication

jensflorian, 2020-03-09:
2020-03-09 22:50:39.133 [WARN ] [ue.internal.handler.HueBridgeHandler] - An unexpected error occurred: null
java.lang.NullPointerException: null
	at org.openhab.binding.hue.internal.State.getAlertMode(State.java:152) ~[bundleFile:?]
	at org.openhab.binding.hue.internal.handler.HueBridgeHandler.isEqual(HueBridgeHandler.java:777) ~[bundleFile:?]
	at org.openhab.binding.hue.internal.handler.HueBridgeHandler.access$6(HueBridgeHandler.java:776) ~[bundleFile:?]
	at org.openhab.binding.hue.internal.handler.HueBridgeHandler$2.doConnectedRun(HueBridgeHandler.java:234) ~[bundleFile:?]
	at org.openhab.binding.hue.internal.handler.HueBridgeHandler$PollingRunnable.run(HueBridgeHandler.java:101) [bundleFile:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:1.8.0_222]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308) [?:1.8.0_222]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180) [?:1.8.0_222]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294) [?:1.8.0_222]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_222]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_222]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_222]

Status of all items then changed from ONLINE to OFFLINE: Verbindung zur Hue Bridge unterbrochen
ConBee II Firmware is 264A0700
Reverting to 2.05.73 reestablishes openhab connection.

TilBumble, 2020-03-10:
Hello.

Similar here:
After updating my ConBee to firmware 26350500 and deconz to  2.05.75 i experience the following:

1. Openhab 2.5.2: Hue Bridge and  eight hue lights toggle between ONLINE and OFFLINE as in opening post. (several seconds online, reconnecting within less than a second)
2. Openhab 2.5.2: deconz-things are marked with "Status: UNINITIALIZED - HANDLER_INITIALIZING_ERROR Invalid URI host: null (authority: 192.168.178.35:80:80)"
yes, two times ":80"
but items work as expected
3. Openhab 2.5.2: Ubisys S1-R disappeared as hue light, but works correctly in Phoscon.
Power and consumption sensors still shown correctly in OH as deconz things
4. Phoscon and API: ConBee still shown as light "Unknown 10"
"swversion": "0x26330500", "type": "Unknown",

5. Phoscon: Gateway-Page still not displayed correctly on Microsoft Edge https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1727#issuecomment-518025105
 (i am not sure if a fix was implied by "Fix bug that gateway settings page did show an offline page.")

I rebooted, page reloaded and reconfigured (OH) several times.

I know: never change a running system. I realized afterwards that the firmware was newer than the one labelled "latest" and seems to be beta.

Thank You all in advance for any hints.

sabefix, 2020-03-11:
> ConBee II Firmware is 264A0700
Reverting to 2.05.73 reestablishes openhab connection.

Same issue and config for me, except I did go back to 2.05.74

mholst, 2020-03-11:
I have the same issue.
@sabefix How did you downgrade? I have just set this up yesterday.

/Morten

lukicsl, 2020-03-12:
Same issue here, I gotta go back to container 2.05.71

FrankH1, 2020-03-14:
Hi there, I'm a Newbie - so please excuse whatever is wrong. 
Im using OpenHab 2 on a Raspi3 and have a Conbee II stick (2.05.75) connected to it.
Phoscon App is installed and works, have so far connected one RGB bulb which is controllable via the Phoscon interface.
Facing Problems in controlling from OH.
Added the Hue Binding to controll the bulb. It seems to be configured but falls over in the frequency of updates. Getting the following in the logfile:

2020-03-13 20:07:32.301 [WARN ] [ue.internal.handler.HueBridgeHandler] - An unexpected error occurred: null
java.lang.NullPointerException: null
at org.openhab.binding.hue.internal.State.getAlertMode(State.java:152) ~[bundleFile:?]
at org.openhab.binding.hue.internal.handler.HueBridgeHandler.isEqual(HueBridgeHandler.java:777) ~[bundleFile:?]
at org.openhab.binding.hue.internal.handler.HueBridgeHandler.access$6(HueBridgeHandler.java:776) ~[bundleFile:?]
at org.openhab.binding.hue.internal.handler.HueBridgeHandler$2.doConnectedRun(HueBridgeHandler.java:234) ~[bundleFile:?]
at org.openhab.binding.hue.internal.handler.HueBridgeHandler$PollingRunnable.run(HueBridgeHandler.java:101) [bundleFile:?]
at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:1.8.0_222]
at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308) [?:1.8.0_222]
at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180) [?:1.8.0_222]
at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294) [?:1.8.0_222]
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_222]
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_222]
at java.lang.Thread.run(Thread.java:748) [?:1.8.0_222]
==> /var/log/openhab2/events.log <==
2020-03-13 20:07:32.317 [hingStatusInfoChangedEvent] - ‘hue:bridge:00212E05564F’ changed from ONLINE to OFFLINE: Verbindung zur Hue Bridge unterbrochen.

2020-03-13 20:07:32.333 [hingStatusInfoChangedEvent] - ‘hue:bridge:00212E05564F’ changed from OFFLINE: Verbindung zur Hue Bridge unterbrochen. to ONLINE

So somehow the update seems to trigger an exception. Nothing configured so far just added it from the PaperUI. I’m lost…
Reading the posts above a rollback of the Conbee soft seems to help. Any suggestions?
Best regards

sabefix, 2020-03-14:
@FrankH1 You need to downgrade deConz to 2.05.74 as in general described here: [https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian](url)
So you need to

- `sudo systemctl stop deconz` (Headless version), or
`sudo systemctl stop deconz-gui` (GUI version)
- `wget http://deconz.dresden-elektronik.de/raspbian/beta/deconz-2.05.74-qt5.deb`
- `sudo dpkg -i deconz-2.05.74-qt5.deb`
- `sudo systemctl start deconz` (Headless version), or
`sudo systemctl start deconz-gui` (GUI version)

FrankH1, 2020-03-15:
@sabefix - that helped - thanks a lot :-)

lukicsl, 2020-03-16:
@manup - is the root cause for that behavior found?

DerOetzi, 2020-03-24:
I think I have found the root cause in the response of {{deconz_endpoint}}/api/{{deconz_user}}/lights of rest call there are two missing attributes inside the ConBee II node at version 2.05.75:

The response of 2.05.74:

```
"24": {
        "etag": "96b435e3e178b027cc8407e9fdda0f2c",
        "hascolor": false,
        "manufacturername": "dresden elektronik",
        "modelid": "ConBee II",
        "name": "Unknown 24",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": "0x264a0700",
        "type": "Unknown",
        "uniqueid": "00:21:2e:ff:ff:04:d9:c3-01"
    },
```

Response of 2.05.75:

```
"24": {
        "etag": "cf49ec8268f7edf7112873b0cba54867",
        "hascolor": false,
        "manufacturername": "dresden elektronik",
        "modelid": "ConBee II",
        "name": "Unknown 24",
        "state": {						
            "reachable": true
        },
        "swversion": "0x264a0700",
        "type": "Unknown",
        "uniqueid": "00:21:2e:ff:ff:04:d9:c3-01"
    }
```

As you can see the two state attributes "alert" and "on" are missing in response.

Maybe that is because of refactoring by @ebaauw to setLightState() or  rest_lights.cpp from Mar 7, 2020. So he can analyze it please.

Best regards

ebaauw, 2020-03-24:
No, this is unrelated to the refactoring, which, I think, hasn’t even been merged yet.  The change between .74 and .75 is to correct #2490.

There have been, in fact, other devices that are exposed as `/lights` resources, but without `state.on`, like the Heiman Siren and the IKEA Trådfri repeater.  These do have `state.alert`, though.  The `type` for the Conbee II should be `"Configuration tool"`, though.
```
{
  "etag": "d4c1b7ecf4e29630d86da2180bce8fff",
  "hascolor": false,
  "manufacturername": "Heiman",
  "modelid": "WarningDevice",
  "name": "Siren",
  "state": {
    "alert": "none",
    "reachable": true
  },
  "swversion": "2017.9.27",
  "type": "Warning device",
  "uniqueid": "00:0d:6f:00:0b:e6:43:b8-01"
}
```
```
{
  "etag": "05932a9fc15a9beb2831bab2f9018f02",
  "hascolor": false,
  "manufacturername": "IKEA of Sweden",
  "modelid": "TRADFRI signal repeater",
  "name": "Trådfri Repeater",
  "state": {
    "alert": "none",
    "reachable": true
  },
  "swversion": "2.2.005",
  "type": "Range extender",
  "uniqueid": "00:0d:6f:ff:fe:27:66:58-01"
}
```

DerOetzi, 2020-03-27:
My pull request on openhab hue-binding to 2.5.4M3 was merged right now. So you can download a version in the next days

jensflorian, 2020-04-20:
OpenHab 2.5.4 is now stable and sucessfully connects with Deconz 2.05.75. Therefore closing this issue.
@DerOetzi Many thanks for your fix!

closed 2020-04-20 by jensflorian