dresden-elektronik/deconz-rest-plugin #2565: After Upgrade to 2.05.75: All devices offline

happenpappen, 2020-03-10:
Hi,

after upgrading my Raspberry Pi 3+ with Raspbee card to deCONZ 2.05.75 all devices (13 Osram Smart+ Plugs and one Aqara Magic Cube) are offline. They are displayed in the phoscon app, but all of them are greyed out and not controllable. Right after the update everything worked fine, the error occurred some hours later. Another thing is, that i had to update the Raspbee firmware with GCFFlasher - the phoscon app didn't realize, that there was a newer one.

I tried to rebind one of the Smart+ plugs, but this does not work. Any hints on this?

kind regards,

Christoph

Tosko4, 2020-03-10:
Can you provide logs? No one can support you this way

happenpappen, 2020-03-10:
This has actually been resolved by the hints given in issue #2566. I would have provided logs, when i've found some... i've checked for options in the phoscon app and under /var/log, but did not find any logs.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix