dresden-elektronik/deconz-rest-plugin #2566: After Upgrade to 2.05.75: Backup not working

happenpappen, 2020-03-10:
Hi,

after upgrading my Raspberry Pi 3+ with Raspbee card to deCONZ 2.05.75 the backup function isn't working anymore. After clicking on "Create backup" the rotating gear appears and after some seconds the message "Creating backup failed" is displayed. The last backup i've created is dated 04.10.2019 so i guess this worked with version 2.05.67 or 2.05.69.

kind regards,

Christoph

manup, 2020-03-10:
I think the backup code hasn't changed in a while. One reason the creation can fail is when the RaspBee isn't shown as connected to deCONZ (therefore the nodes aren't shown);

Since you experience the devices gone offline you may check the Zigbee configuration page to see if any parameters have changed recently. Here you can also revert to former working parameters.

Here is a description how it works:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

happenpappen, 2020-03-10:
Thanks for your feedback - on the given (hidden) configuration page i could see that the channel as well as the panid had changed from the last version to the current. Are they supposed to be changed due to an update? The  list shows the same channel and panid pair since the first version which is recorded as 2.05.58... after loading the values recorded for version 2.05.73 the devices came back online and the backup options works again.

BTW: On Mac OS you must use Shift+Option+Click to get to the hidden page - maybe you could add this to the page?

manup, 2020-03-10:
Glad it worked. The channel and PANID shouldn't change after an update, but this is currently not checked. Might be an good idea to add this to the process so that the configuration gets verified automatically.

>BTW: On Mac OS you must use Shift+Option+Click to get to the hidden page - maybe you could add this to the page?

Good hint, we should add this to the FAQ or Wiki section.

SwoopX, 2020-03-10:
> Good hint, we should add this to the FAQ or Wiki section.

@manup Already took the liberty a while ago ;)

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix