dresden-elektronik/deconz-rest-plugin #2567: Issue with Xiaomi Aqara Motion sensor

Maddinthebrain, 2020-03-10:
Hi, 
I'm using a Xiaomi Aqara Motion Sensor. Together with FHEM it switches a Light device per MQTT. Over all it works as it should. But, every 54mins the light is switched on without motion recognized. The FHEM log file contains the following:

2020.03.10 09:22:49 4: PIR_Flur_UG: lastupdated: , hash->{lastupdated}: 2020-03-10 07:27:47, lastupdated_local: , offsetUTC: 0
2020.03.10 09:22:49 4: parse status message for PIR_Flur_UG
2020.03.10 09:22:49 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 09:22:49 4: PIR_Flur_UG: lastupdated: 2020-03-10 08:22:49, 
hash->{lastupdated}: 2020-03-10 07:27:47, lastupdated_local: 2020-03-10 
09:22:49, offsetUTC: 3600
2020.03.10 09:22:49 3: MQTT2_DEVICE set FlurUG_Deckenlicht on
2020.03.10 09:26:18 4: parse status message for PIR_Flur_UG
2020.03.10 09:26:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 09:31:18 4: parse status message for PIR_Flur_UG
2020.03.10 09:31:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 09:37:18 4: parse status message for PIR_Flur_UG
2020.03.10 09:37:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 09:43:18 4: parse status message for PIR_Flur_UG
2020.03.10 09:43:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 09:49:18 4: parse status message for PIR_Flur_UG
2020.03.10 09:49:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 09:55:18 4: parse status message for PIR_Flur_UG
2020.03.10 09:55:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 10:01:18 4: parse status message for PIR_Flur_UG
2020.03.10 10:01:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 10:07:18 4: parse status message for PIR_Flur_UG
2020.03.10 10:07:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 10:13:18 4: parse status message for PIR_Flur_UG
2020.03.10 10:13:18 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 10:17:50 4: parse status message for PIR_Flur_UG
2020.03.10 10:17:50 4: PIR_Flur_UG: lastupdated: , hash->{lastupdated}: 
2020-03-10 08:22:49, lastupdated_local: , offsetUTC: 0
2020.03.10 10:17:50 4: parse status message for PIR_Flur_UG
2020.03.10 10:17:50 4: PIR_Flur_UG: use offsetUTC 3600 from bridge
2020.03.10 10:17:50 4: PIR_Flur_UG: lastupdated: 2020-03-10 09:17:50, 
hash->{lastupdated}: 2020-03-10 08:22:49, lastupdated_local: 2020-03-10 
10:17:50, offsetUTC: 3600
2020.03.10 10:17:50 3: MQTT2_DEVICE set FlurUG_Deckenlicht on

It is obvious that after a entry "PIR_Flur_UG: lastupdated: 2020-03-10 08:22:49, 
hash->{lastupdated}:" the light is switched on every time. This happens every 54min. As the sensors keeps the motion status at least for minutes and in FHEM the status remains at visibel at nomotion it must be at motion for a very short period of time, so it triggers the light. And as there braces it may be come from the REST API. I'm using deconz V2_05_74.

Thanks Martin

SwoopX, 2020-03-10:
Hi, that's not a deconz issue, it is related to your notify as far as I can tell. The Xiaomi devices update all their values every ~50 Mins or so.

Are you triggering on a state change? You may want to add event-on-change-reading.

Maddinthebrain, 2020-03-10:
Thanks, that is the solution.

closed 2020-03-10 by Maddinthebrain