dresden-elektronik/deconz-rest-plugin #2568: RaspBee on Pi 3B+ no connection to any Hue-App

coke7, 2020-03-10:
I'm running Hass.io on a Pi 3B+ with a Raspbee.
When I first got the system up, it worked with the original hue-app as well as with all 4 hue.
Since I changed some things (don't know which any more) ans re-set up the whole system again it doesn't find any of these apps. Works great within the phoscon app.
Maybe the problem came from changing to https!?
Perhaps the later installed pihole blocks anything important..???
Any ideas?

thanks in advance
Marc

Mimiix, 2020-06-06:
I think this has to do with UPNP in the HA container, not reaching it.

closed 2020-06-12 by Mimiix

coke7, 2020-10-25:
Problem solved: changed the gateway name to one without a special character. Then hue essentials found the bridge right away

Mimiix, 2020-10-26:
Thats a nice one. @coke7 , can you try to rename it again and see what happens? 

I'll tag @Thomas-Vos  along.

Thomas-Vos, 2020-10-26:
If it fails again, could you share the exact name of your gateway which did not work in Hue Essentials?