dresden-elektronik/deconz-rest-plugin #2569: [Q] RasBee II with dockerized deCONZ. RTC and Migration?

stickpin, 2020-03-10:
Hi All,

I've purchased RasBee II today and I see that one of the features there is RTC.
The question, should it or will it work with dockerized deCONZ or if it's needed at all in the dockerized environment?

and the second question, if I want to migrate from ConBee II to RasBee II, what should be the process?

Thanks in advance!

LukasA83, 2020-03-14:
Tried the same. I restored a backup and see all devices in VNC. But no device is reachable. If I reset on device and connect it via Phoscon, it works flawless.
Network setting look 100% same.

Setup is:
Raspberry Pi4b
Docker marthoc image 2.05.74 (having issues with .75 and openhab)

Do I need to "reconnect" all devices? I thought restoring the settings would clone the conbee2 to Rasbee2 so that it works 1:1.

manup, 2020-03-14:
Hi currently we have seen a issue that the backup from RaspBee I and restore to RaspBee II didn't work smoothly (as it should) we are investigating for a fix. In our test it took 20 minutes after the restore before the devices became reachable again.

One thing that might meanwhile help is the Zigbee configuration page which can apply older settings back to the device. Its usage is described in:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

LukasA83, 2020-03-14:
ok, switched docker again to rasbee2 (by the way migrated from conbee2) and will let it running for a while. Once I'm back, I'll post an update...

In the hidden configuration page, PANID and network key are equal over all versions.

stickpin, 2020-03-14:
@LukasA83, @manup in my case nothing helped, spend quite some time by trying to make it work.
After a few hours of trying I gave up on RaspBee2 and returned back to ConBee2.
On the way, I also gave up on Raspberry 4b for now as it creates a lot on interference once something connected to USB3 port (SSD drive for example) and Zigbee network becomes unstable.

manup, 2020-03-14:
I think one reason why devices might not connect directly after the transfer is the low level NWK security counter which is not transferred currently. One way to check if that is the case is to powercycle a light and see if it becomes responsive.

stickpin, 2020-03-14:
@manup have tried it also, didn't work for me. Maybe @LukasA83 will have more luck.

LukasA83, 2020-03-14:
@manup after 30 minutes still the same. Tried to power cycle a plug and no update. I see these messages in the log when trying to read a cluster:
 

12:30:10:060 max transmit errors for node 0x00158D000388EDEC, last seen by neighbors 418 s,

timbru31, 2020-04-23:
Experiencing the same issue :( - I've switched from a RaspBee I module to a ConBee II, after importing the backup all lights were marked as available but did not respond, now way after 30 minutes all lights are greyed out/marked as unavailable.  
My old Raspberry Pi is completely shutdown with the power cord removed.

Anything I can assist to debug this issue? Unfortunately, that's a very, very bad customer experience because it's even advertised as a working scenario in the wiki.

I'll leave my network running in this state for some hours and will report back.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

alekslyse, 2020-11-14:
Not sure if migration TO rasbee2 is completely solved. It works conbee 2 -> rasbee, rasbee -> conbee2, but not conbee2/rasbee -> rasbee2

I setup a RP4 with the premade image. Started it first with rasbee (1) and restored the backup from HA docker image (conbee 2) - worked fine no issues. Swapped conbee 1 with 2 and restored the same image. It start to mesh, but not much. Tried to let it stay for an hour and still cant turn on a light. Loaded up the conbee2 again and it meshes is seconds (conbee2 and rasbee2 got same fw). 

Here is some of the log from the rasbee2 and it seems like its all working ok? https://www.codepile.net/pile/K17lqPnA  - it just seems like rasbee2 do not want to work with the backup that work on rasbee1 and conbee2

alekslyse, 2020-11-14:
Just to try I just put the conbee2 into the rpi and booted it, again with same fw as the rasbee2 and the mesh was created right away