dresden-elektronik/deconz-rest-plugin #2570: Xiaomi motion, not detected state, takes 5 min instead of 2

EduardSarjadjuk, 2020-03-10:
I have to reopen issue 2523.

This was the problem from pergolafabio:
Normally , my xiaomi motion sensors are in “not detected” state after 2 minutes of no movement… Now it takes 5 … Also in phoscon I see movement detected… What can be the cause? Battery failing? It’s only one sensor behaving like that, others are fine…
It’s still reporting 100% as battery state…

It was recommended to delete the sensors and reconnect them.
I did that, but
after a certain time, switch back to 5min mode. Deconz (.74 & .75)

sorry for my bad englisch.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2523

salopette, 2020-03-11:
I have also noticed this problem since the last update.

EduardSarjadjuk, 2020-03-11:
the problem does not exist in version .73.  O habe the problem with Version .74 and .75.

Smanar, 2020-03-11:
> 
> 
> the problem does not exist in version .73. I have the problem with Version .74 and .75.

So you are thinking it's from the API and not the device ?

EduardSarjadjuk, 2020-03-11:
i think it's the API.  I used two new sensors to try that out.  The error also occurred with these new sensors.  After downgrading to .73 everything works again.

salopette, 2020-03-11:
I also think it's API because it happens to several people.

[image]

mkarnebeek, 2020-03-13:
I also noticed the longer timeout value since .74

Smanar, 2020-03-13:
Ok, so same for me
1 sensor with 300 duration > 5 mn
1 sensor with 90 duration > 5 mn too

Lol, I m checking the code, I don't see where it can come.

pergolafabio, 2020-03-30:
ok, my problem is back, i created this issue here : 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2523
i fixed it by just deletering and repairing that sensor

now i notice since several days another sensor is impacted 

the same issue issue again, motion to OFF detected only after 5 min instead of 2 

how can we resolv this without repairing again ?

pergolafabio, 2020-04-04:
Can this thread be closed? The other one is open again, so it's a duplicate

EduardSarjadjuk, 2020-04-05:
yes i close it

closed 2020-04-05 by EduardSarjadjuk