dresden-elektronik/deconz-rest-plugin #2573: Friends of Hue questions

bastshoes, 2020-03-10:
I've got FOH switch by Senic.
Paired without issues. Buttons sends events. I have some questions regarding this type of switches. 
1. Switch didn't appear in GUI (on the map). Is it expected? 
2. Device group wasn't created for the switch. Is it expected?
3. How to create group binding with FOH switch and lights?

ebaauw, 2020-03-10:
Yes, yes, not.

Long answers:
1. ZGP devices don't show in the GUI, as they don't expose any descriptors, endpoints, nor clusters;
2. ZGP devices cannot control lights directly.  They send very short ZGP messages, which need to be picked up by a nearby ZGP Proxy (the coordinator or a router with a 0xF2 _Green Power_ endpoint and 0x0021 _Green Power_ clusters).
3. You need (an app) to create deCONZ rules to control the group/lights from the button events.  Note that the switch will work only when in direct range of the coordinator.  Unlike the Hue bridge, the REST API plugin doesn't (yet) setup the ZGP Proxies to forward the ZGP messages to the coordinator over the mesh network.

bastshoes, 2020-03-10:
Thank you Erik for clarification.

closed 2020-03-11 by bastshoes

jensflorian, 2020-04-28:
I hope you don't mind me asking further questions here:

Ad 2: Is it possible in Deconz GUI to read from a ZGP proxy (or to be specific from the Green Power cluster) whether they are the destination of a switch? The GP basic proxies of the Philips plug seem to be empty, probably because they haven’t been configured yet. If I want to configure these to forward ZGP messages I’d need a hue bridge?

Ad 3: So unlike ZLL, ZGP rules work won’t without a running coordinator and ZGP Switches can’t be placed outside the ConBee stick range?

jaxer, 2020-07-24:
Also interested. I have about 70 nodes, mostly hue lights with GP clusters all over the house, but ZGP switches (senic enocean based) only work reliably when placed in same room as conbee II stick. in other rooms packets are getting missed, you have to click switch many times in a row.
Is there a way to extend range of GP devices somehow?

dusopn, 2021-07-26:
> I hope you don't mind me asking further questions here:
>
> Ad 2: Is it possible in Deconz GUI to read from a ZGP proxy (or to be specific from the Green Power cluster) whether they are the destination [...]

Hello, I would also like to know answers to these questions.

BR,
Dusan