dresden-elektronik/deconz-rest-plugin #2574: Cobee2 / xiaomi door sensors / deconz logs

SAOPP, 2020-03-11:
Hi!

Guys just next question. I think I faced with 50 nodes of my raspbee, and start to repairing some devices to my conbee2 of another instance...

First I had deleted from HA/Deconz addon of my rapsbee two contact sensor of xiaomi (door sensors), deleted it, and reboot the raspbee addon, after this I start to pairing this devices to the my condee2, all work fine, but, I open the container logs of my conbee, and see next lines:

```
10:18:39:147 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
,10:49:39:147 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
,11:00:51:146 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)

```
It also seemed to me that this happens after I used the door... and after this these logs lines are showed in logs.

And also saw this:
```
10:18:39:147 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
,10:49:39:147 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
,11:00:51:146 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
,11:10:39:438 Announced to internet https://phoscon.de/discover
,11:10:49:984 no button map for: lumi.sensor_magnet.aq2 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
,11:10:46:795 no button map for: lumi.sensor_magnet.aq2 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
,11:10:46:795 ZCL attribute report 0x00158D00034683AA for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
,11:10:48:472 no button map for: lumi.sensor_magnet.aq2 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
,11:10:48:472 ZCL attribute report 0x00158D00034683AA for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
,11:10:49:985 ZCL attribute report 0x00158D00033A72DE for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
,11:10:52:434 don't close database yet, keep open for 900 seconds
,11:10:52:440 no button map for: lumi.sensor_magnet.aq2 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
,11:10:52:440 ZCL attribute report 0x00158D00033A72DE for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
,11:10:55:227 GW firmware version: 0x264a0700
,11:10:55:227 GW firmware version is up to date: 0x264a0700
,11:10:57:147 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
,11:10:59:146 skip binding for attribute reporting of ep: 0x00 cluster 0x0006 (end-device might sleep)
```
Only two sensors right now connected to the conbee2, xiaomi aqara door sensors. And seems work fine, but what about this logs? Any suggestions?

SAOPP, 2020-03-11:
And I’ll supplement my post a little.

The first screenshot from my main raspbee, I deleted the sensors from there, two sensors that I connected to conbee2.
[image]

...and below my screenshot of conbee2:
[image]
As u can see, I have maybe problem with bathroom contact sensor? 'Coz at first screenshot we can see some spirit of this sensor at respbee.

SAOPP, 2020-03-11:
`15:22:42:146 send permit join, duration: 0
`

[image]

Just watching the logs, and had seen also this. :) It's just for information. I'm not at home right now. Any comments guys?

I’m just thinking that probably when deleting devices from one coordinator, it is necessary to turn off the power, if I'm going to pair these devices to another coordinator, am I right? I just want to understand the correct procedure if there are more than 1 coordinator in use.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix