dresden-elektronik/deconz-rest-plugin #2575: 2.05.75 Cannot pair (Ikea) bulbs, no contact with paired bulb

oyta, 2020-03-11:
Hi! I ran into some exhausting problems, and I cannot get any further.
I cannot pair (IKEA) light bulbs. 
I can pair Xiaomi temp sensors and water sensors. I can also pair IKEA switch (with 5 buttons).

HW: RPI 3B with Raspbee
OS: Latest Raspbian (installed last week).

I had the .74 release with the HW showing as a light. Since I only had a few devices paired, I thought I would try to fix it. The changeling said it was fixed in the 75-release, so I:

1. Manually updated the FW and Deconz to latest to remove it (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually). It didn't go away, so I:
2. Reset the GW through Phoscon.
3. Paired a light bulb, went ok.
4. Could not pair a second bulb, nor any others. Tried several different bulbs, but no response. They're IKEA, and I am able to reset them, however nothing happens in the Phoscon GUI.
5. I've lost contact with the one bulb that I did manage to pair with.
6. In between, able to pair with different sensors and an Ikea Switch.

[image]

`journalctl -u deconz`

> -- Logs begin at Wed 2020-03-11 22:17:01 CET, end at Thu 2020-03-12 00:08:09 CET. --
> Mar 11 23:07:57 raspberrypi systemd[1]: /lib/systemd/system/deconz.service:8: Unknown lvalue 'StartLimitIntervalSec' in section 'Service', ignoring
> Mar 11 23:07:57 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
> Mar 11 23:07:57 raspberrypi deCONZ[416]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
> Mar 11 23:07:57 raspberrypi deCONZ[416]: libpng warning: iCCP: known incorrect sRGB profile
> Mar 11 23:07:58 raspberrypi deCONZ[416]: This plugin does not support propagateSizeHints()
> Mar 11 23:07:58 raspberrypi deCONZ[416]: This plugin does not support propagateSizeHints()
> Mar 11 23:08:08 raspberrypi deCONZ[416]: This plugin does not support propagateSizeHints()
> 

[image]

LitenGolvLampa is the bulb I managed to pair, but not able to connect to no longer.
[image]

I tried to find better Deconz logs, but couldn't. Where are they placed?

Any recommended steps in order to fix this problem?
Any additional information I can provide to ease debugging?
Any debugging I can try myself?

oyta, 2020-03-12:
Thought I would add some more info here. Trying to get it work, but no such luck. 

I started Deconz with debug logging as follows:
`deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80 > deconz.log 2>&1
`
See the complete log here: [deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4326049/deconz.log)

I did not find anything particular, but did not know what to look for either.

According to your [wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Pairing-Devices) the controller is only able to have between 10 and less than 32 end devices.

> As of firmware 0x26240500, they support up to 32 end devices; older firmware versions only support up to 10. EDIT It's fewer again with later firmware versions (due to memory restrictions), but I don't know how many.

I have ten devices. Even though I have not paired the 0xF912, 0xFF92 and 0xD4ED (and I am not sure what it is - they are not visible in Phoscon). Could that be the reason why I am not making this work? Still no contact with the paired Ikea bulb though, evne if its 2 meters away from the RPi+Raspbee.

oyta, 2020-03-12:
Actually, reading a bit more I found a [wiki page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log) describing some error codes from the logs.

I found one of them, with only one occurrence, on line 108 in the log file I posted in the last post:
`18:35:29:596 APS-DATA.confirm id: 8, status: 0xD0 `

Explanation does not match:
> 0xD0 No route to the receiver exists
> This appears usually when the receiver is powered off and is issued after a series of 0xE9 or 0xA7 codes. Note that routing is figured out and repaired automatically by all routing device.

 I don't have E9 or A7 codes. I am not sure what this means.

Homey1996, 2020-03-13:
I have the exact problem with pairing the Ikea bulb (same firmware as you).
I have a new setup but just cant pair the ikea bulb, i sometimes get a bulb detection but it says unknown and i cant control that bulb.

I hope this bug get fixed soon.

gareth-ellis, 2020-04-15:
I'm having the same issue, I originally could pair but not control the bulb....now I'm struggling to even pair it

Mimiix, 2020-06-05:
Is this still a issue?

gareth-ellis, 2020-06-05:
After a lot of messing about I got them paired - used the reset function in the old UI then eventually it worked

Mimiix, 2020-06-05:
@gareth-ellis Case closed i guess! Thanks for the reply.

closed 2020-06-05 by Mimiix