dresden-elektronik/deconz-rest-plugin #2576: [Device Support Request] Samsung SmartPlug 2018 Model ID: GP-U999SJVLDAA

jbruening, 2020-03-12:
https://www.samsung.com/us/smart-home/smartthings/outlets/samsung-smartthings-outlet-2018-gp-u999sjvldaa/

Side of the unit also lists M/N "IM6001-OTP01"

This looks pretty similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2121, but lacks the 'Simple Metering' cluster

Phoscon currently treats this as an on/off "light". It supposedly supports power monitoring as well though.

[image]

[image]

[image]

Electrical measurement cluster picture taken with a 20 inch box fan set to slowest setting of 3
[image]

SwoopX, 2020-03-12:
I can take care of that. Can you please provide a screenshot of the node info panel as well? Thanks!

jbruening, 2020-03-12:
[image]

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

jbruening, 2020-06-11:
Yes, it is resolved. I re-added the device in deconz (2.5.75) (which appeared to show nothing different in phoscon about it), and then restarted home assistant.  There is now a wattage sensor showing up.

[image]