dresden-elektronik/deconz-rest-plugin #2577: v2.05.75 can't remove gateway as light source

docholiday, 2020-03-12:
Hi there,

first of all, I have to say you guys are great! I love the RaspBee, deCONZ and the phoscon app :-)

Small problem though: As you know, version 2.05.74 brought the gateway as a new light source, which was a bit irritating (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2519).
According to the changelog for v2.0.5.75 this should be fixed and if one still sees the gateway you just should remove the light source.

However, after the update the gateway was still listed as light source AND I can't remove it:
Whenever I try to, the light is shown as removed, but phoscon app seems to crash afterwards as when I click on the arrow to get back to the main window, I receive the message 

> 502 Bad Gateway

After a restart of the deconz add-on the gateway is listed again as light source. I tried this quite a few times, so it is reproducible (at least on my setup).

My system:
Hassio on Raspberry Pi 3B
HassOS 3.12
Home Assistant 0.106.6
Supervisor 209
deCONZ add-on 5.3.2
RaspBee I with firmware 26330500

P.S.:
In the deCONZ addd-on log I always see the following message on startup:

> discovery network reply error: Error transferring http://dresden-light.appspot.com/discover - server replied: Service Unavailable

MBeeken73, 2020-03-12:
Same behavior with my setup!

schwatter, 2020-03-12:
Look here

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2534

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix