dresden-elektronik/deconz-rest-plugin #2578: innr SP120 Smartplug not appearing under 'Sensors' in Phoscon App

martingruening, 2020-03-12:
I bought a couple of of innr SP120 Smartplugs. They were easy to add to deCONZ via Phoscon App (Conbee II, 2.05.74, Firmware 264A0700). After adding they appear under lights in Phoscon with the Wallplug icon (so I guess you can switch them on/off, not tried). But I expected them also to appear under sensors as each plug represents two sensors. I guess not appearing as sensors is a bug.

Through the REST API they work fine (which is fortunately my main use case):

/sensors
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "98943ec8fca64d3b2b56c3f0c2d5eabb",
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "SP 120 (5)",
    "state": {
        "consumption": 1270,
        "lastupdated": "2020-03-12T06:50:45"
    },
    "swversion": "2.0",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:03:89:18:04-01-0702"
}
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "14893ab2d0e2c734405bcbb72512702c",
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "SP 120 (5)",
    "state": {
        "current": 92,
        "lastupdated": "2020-03-12T06:51:09",
        "power": 6,
        "voltage": 226
    },
    "swversion": "2.0",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:03:89:18:04-01-0b04"
}

/lights
{
    "etag": "7580cc9d54731f1ad8be8fe37d60ac07",
    "hascolor": false,
    "manufacturername": "innr",
    "modelid": "SP 120",
    "name": "TV",
    "state": {
        "alert": "none",
        "on": true,
        "reachable": true
    },
    "swversion": "2.0",
    "type": "On/Off plug-in unit",
    "uniqueid": "00:15:8d:00:03:89:18:04-01"
}

SwoopX, 2020-03-12:
No bug, but expected behavior. Phoscon is closed source and unless it is not explicitly added by DE, new sensors may not show up.

YKO-de, 2020-03-12:
That's correct. We always indicate such plugs as a device. Regardless of the sensors in the plug.

hwinkel, 2020-05-05:
What does this mean, the sensor is not available? Just ordered a few conbees and have a bunch of SP120 just for the reason to also see consumption, as we was under the impression that SP120 is supported with sensor.

SwoopX, 2020-05-05:
@hwinkel It's working, the issue is about visibility in Phoscon, which is more cosmetics.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix