dresden-elektronik/deconz-rest-plugin #2579: ConbeeII stick is shown as a Raspbee???

fribse, 2020-03-12:
The deconz finds the zigbee device without problems, but misidentifies it.
This is getting worse over the last few versions.

The log looks like this:

`
22:50:15:906 COM: --dev: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2139528-if00 (RaspBee)
`
The conbeeII stick is installed in a RPi4 with Raspbian and Home Assistant running with the deconz plugin.

Deconz: 2.05.75
FW: 264A0700

SwoopX, 2020-03-12:
Is that a problem in any way? I also observed it, but don't really mind since it doesn't seem to cause any harm.

fribse, 2020-03-14:
Well, that is what I'm not sure off. Last time I rebooted the machine, it started by showing it as a conbeeII, and then changed it's mind to a raspbee.
In any case, for debugging for the non-primary maintainer of a home setup, it's quite confusing.

mbo18, 2020-03-14:
I have the same issue since I have updated to .74 or .75.
Here are my logs:

> 11:12:40:503 COM: --dev: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2133629-if00 (RaspBee)
> 11:12:40:558 Device firmware version 0x264A0700

I'm using HA (hassio) and uninstall/re-install of Deconz add-on did not solved the issue

lostkeys, 2020-03-18:
+1. Just updated to 2.05.75 and my conbee 2 stick shows up as raspbee and nothing works 😕
I got the same logs as mbo18 above...

lostkeys, 2020-03-18:
After two hours of randomly clicking, downgrading, upgrading and restarting Home Assistant my Ikea remotes finally started working. But the Conbee 2 Stick still shows up as Raspbee

ZyberSE, 2020-04-05:
Same for me.

fribse, 2020-04-17:
I've converted the HA to a Proxmox-on-nuc thing, and in the HA on the nuc it is shown as a conbee stick again.

schumi2004, 2020-04-26:
Same issue here which started to happen after i needed to reboot my HassIO/RPi setup.

mbo18, 2020-05-01:
Had to reinstall HassOS 3.13 from scratch due to another issue. After having restored my backup, my Conbee II is again recognized correctly as a Conbee II

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

mbo18, 2020-06-06:
Still active. If I use /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2133629-if00 it is recognize as a raspbee. If I use /dev/ttyACM0 it is recognize as ConbeeII

mbo18, 2020-06-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1221

Mimiix, 2020-06-06:
It has been known for a bit. I think this is a lead. Keeping it open and assigning a To-Do.

JapieJoo, 2020-07-13:
Same issue over here. Conbee II stick shown as Rasbee. Firmware upgrade starts (via addon) says its finished but stucks on same version.

Similar to above: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2133629-if00 changed to  /dev/ttyACM0 and..... tadaa, stick is correctly shown as Conbee II usb. Could perform a firmware upgrade after that. Addon said it finished, but no...

HassOS 4.11
Deconz addon:  5.3.6

Smanar, 2020-07-13:
IDK how you do, but the firmware upgrade using Phoscon in Hassio not working, need to use another machine.

Mimiix, 2020-07-13:
This is something HA needs to fix on their end, as it is fine on Native deCONZ installs.

kedante, 2020-07-19:
Same problem, "fixed" with "/dev/ttyACM0", but I don't think it's a definitive solution :(.

deCONZ 5.3.6
HassOS 4.11
Conbee II firmware: 26580700 (update with windows)

Mimiix, 2020-08-10:
I've got confirmation of Manup that he will adress this. Not sure if this will be up in next version.

closed 2020-09-07 by stale[bot]

MurphyB17, 2022-07-10:
Hi, I am having the same issue. I am running on a Raspberry Pi through Homebridge and the homebridge-deconz plugin, I tried reimaging the SD card to with the homebridge image, and reinstalled deconz and the homebridge-deconz plugin to no help. I can't even click on the gateway through the phoscon app. 

I'm a newbie, how can I change the USB location?

[image]

[image]