dresden-elektronik/deconz-rest-plugin #2580: Cant deactivate Access Point

fitim7, 2020-03-13:
Hi, 
like the title says I can't deactivate the access point. I'm using a Raspbee with Raspbian connected via lan. The raspberry is used just for deconz.
The problem is that deconz is always setting up a wifi network called "Phoscon-Gateway-XXXX", wich is regular if you set it up to work as access point. But why can't I deactivate that. I don't need it and don't want deconz to work as access point.
"sudo systemctl stop deconz-wifi" stops the network for 5 seconds, but It pops up again then.
"sudo systemctl disable deconz-wifi" isn't doing anything...
If I try to stop it directly from the Phoscon UI, it says "
"Deactivating access point.
This may take a few seconds." ... but this doesnt finish.

Please help, since this is really annoying for me.

jan666, 2020-03-13:
I had some issues with the access point on my raspberry pi, too. As it's connected via ethernet I disabled wifi completely in the end. Not a real fix but a good workaround for me.

I did that by adding

```
# disable WiFi
dtoverlay=disable-wifi
```

to `/boot/config.txt`

But be careful if you need wifi to access your raspberry pi. Won't work until you remove that line again.

Smanar, 2020-03-13:
Another method https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2387

fitim7, 2020-03-13:
> Perhaps another method #2387

I don't see how this is a method? I don't want to use neither client mode nor access point mode. Deconz should just use the lan network for internet connection.

fitim7, 2020-03-13:
> I had some issues with the access point on my raspberry pi, too. As it's connected via ethernet I disabled wifi completely in the end. Not a real fix but a good workaround for me.
> 
> I did that by adding
> 
> ```
> # disable WiFi
> dtoverlay=disable-wifi
> ```
> 
> to `/boot/config.txt` (and rebooted)
> 
> But be careful if you need wifi to access your raspberry pi. Won't work until you remove that line again.

Thanks! But like you already said: not a real fix but a good workaround.
For me thats a good solution.... but deconz should still fix this issue.

Mimiix, 2020-06-06:
This is a Phoscon issue and not a deconz issue. Will be moved Repo's.