dresden-elektronik/deconz-rest-plugin #2582: Hue Play HDMI Sync

CDRX2, 2020-03-13:
Anybody tried this with Deconz / Raspbee? Would be interested in feedback if it works or not.

wvuyk, 2020-03-13:
Hue Play HDMI Sync has its own API, not conforming the Hue API. I have it working in my plugin, but it would never 'connect' to any other Zigbee controller. It is not Zigbee, but WIFI communication you need to use here

CDRX2, 2020-03-14:
Thanks for the clarification. But the lights are still piloted via Zigbee, are they not? So somehow the Hue Play box has to talk to the Hue gateway, and if I understand you correctly this happens via a specific API, correct?

ebaauw, 2020-03-14:
The sync box talks to the lights over the ZigBee network alright, but in a non-standard way.  I don't think support by any other manufacturer is likely.

Basically the flow is as follows:
1. Hue API client (app) sets up entertainment zone on the Hue bridge using the (REST) [Hue API](https://developers.meethue.com/develop/hue-api/). 
2. Hue Play HDMI Sync box client (app) sets up syncing on the Sync box using the (REST) [Hue entertainment API](https://developers.meethue.com/develop/hue-entertainment/hue-hdmi-sync-box-api);
3. The Sync box sends commands to the Hue bridge the IP network, using the UDP-based [Streaming API](https://developers.meethue.com/develop/hue-entertainment/philips-hue-entertainment-api/#getting-started-with-streaming-api).  The [Hue Sync app](https://www2.meethue.com/en-hk/entertainment/hue-sync) uses the same Streaming API;
4. The Hue bridge sends manufacturer-specific commands to the designated proxy light in the entertainment zone, using ZigBee unicast messages.  These commands haven't been documented, but I suppose they could be reverse engineered using a ZigBee sniffer;
5. The proxy light forwards these manufacturer-specific commands to the other lights in the entertainment zone, using MAC level broadcasts.  By using this unicast/MAC broadcast combi, they claim to reach up to 10 lights with 25 streaming commands per second, without choking the ZigBee network.

So for deCONZ, or any other gateway to emulate the Hue bridge behaviour, it needs to implement:
- The REST API extensions to setup the entertainment zone and start or stop the streaming (1);
- The server side of the Streaming API (3); and
- The client side of the manufacturer-specific ZigBee commands (4).

For any light to be used in an entertainment zone, it needs firmware support for the manufacturer-specific commands (5).  At least one light needs firmware support to act as a proxy (4).

CDRX2, 2020-03-15:
Thanks @ebaauw  for the thorough info. Will therefore leave the project be...

closed 2020-03-15 by CDRX2