dresden-elektronik/deconz-rest-plugin #2584: All sensors takes a while to be reachable .

MattL0, 2020-03-13:
Or i need to mannually activate them by passing in front of them.. or  closing doors etc.. 

I am on deconz .73 . Was on .75 when i saw this... 

I can't remember what build didn't have this problem.

MattL0, 2020-03-13:
what is the last release? do not want to be on beta

Smanar, 2020-03-14:
You mean after a restart ?
Or all the time, when you have triger one, it stay reachable for a long time ?

MattL0, 2020-03-14:
After restarting deconz only.

Smanar, 2020-03-14:
Ok so it's something normal.
For exemple Xiaomi device send notification only every hour or if you have a changement, the rest of time they are sleeping.
So if the device have already send its notification and nothing change for him, the conbee need 1 hour for the next notification and set the device in reachable state.

Sensor can send request, but they don't answer to the gateway.

MattL0, 2020-03-14:
Ha ok thanks a lot!  So the xiaomi sensors will stay in the network, even if they are unreachable since one hour?

MattL0, 2020-03-14:
Will have to modify some automation here. But it is weird that it was no behaving like that before... the last time I saw this was in the .35-50 deconz version

Smanar, 2020-03-14:
Warning, I m say that, just if you restart deconz.
You will not have this problem if deconz stay online.

>Or i need to mannually activate them by passing in front of them.. or closing doors etc..

If the device was realy out of the network, you will not see it when you activate them using them.

MattL0, 2020-03-14:
Then deconz should mark them as reachable ...and if there is no sign of then after 2 hours then it should set them as unreachable .

Sent from an Iphone

Smanar, 2020-03-15:
Only 2 hours ?
I was sure it was at least 24h.

ebaauw, 2020-03-15:
`reachable` is utterly useless and best ignored.  See #2590.

MattL0, 2020-03-15:
> Only 2 hours ?
> I was sure it was at least 24h.
> 
> BTW.
> Are you looking for reachable state on phoscon or on third application ?
> 
> Because I have just make a fast try (reboot deconz), and reachable state havent moved on Phoscon. But was realy short off line.

I am using it on homeseer (via norered deconz ) . And my sensors becomes unreacheable. 

I will just remove the automation that use the unreachable state to send me a pushover message . Thanks guys. 

Hope it will be removed for last seen !

closed 2020-03-15 by MattL0

MattL0, 2020-03-15:
Didn,t a little test today because this time the senors where not unreacheable ( deconz v.72) at restart of deconz in phoscon

- on restart of deconz sensors are not unreacheable
-on restart of computer sensors are not unreacheable 
-updating to deconz v.2.05.75 : sensors are not unreachable. 

So.. i guess it was some sort of ''healing'' or just a coincidence yesterday?

Smanar, 2020-03-15:
Coincidence for me ^^, I think it's a timing situation.