dresden-elektronik/deconz-rest-plugin #2586: Will any of those zigbee to IR work?

ivanpianobcn, 2020-03-14:
Hello everyone, and thanks in advanced for taking some time trying to help me.

After looking for a while, I found there are some “zigbee to IR remote” products apparently accessible for buying. I’m not sure if any one of those will work on Home Assistant, and using deconz.

I make a list here:

https://zigbeealliance.org/zigbee_products/ir-convertor/

http://www.zigbee.com.my/products/en/AV-Controls/IR-Infrared-Emitters/IR-Infrared-Emitters-Full-Directions-p80c63c78.html

https://www.se.com/sg/en/product/CCT501401/wiser-360degree-zb-ir-convertor/

Of course, I know the star on IR connectivity is the BroadLink Rm Pro, or the similar ones but I really want to keep everything into zigbee technology…

Will any of those will be easy to integrate with deconz, and then with Home Assistant?

Thanks!

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

Red-M, 2021-06-14:
I'd love to see these working