dresden-elektronik/deconz-rest-plugin #2587: Aquara motion senzor - Lux Meter

elpaso666, 2020-03-14:
Hello,

the motion senzor does not send update of Lux Meter value, the MSG object has following attributes

[image]

but the LUX value is not being send, can you please add it?
How the treshold can be modified from Phoscon app (I am running liteweight version of Conbee).

thx
Jakub

Smanar, 2020-03-14:
Hello,
The aquara motion sensor create 2 sensors, 1 for presence detection and one for lux.

elpaso666, 2020-03-14:
Correct but the Lux senson is not having the Lux value as you can see in my 1st post

Smanar, 2020-03-14:
```
{
  "config": {
    "battery": 100,
    "on": true,
    "reachable": true,
    "temperature": 2200,
    "tholddark": 12000,
    "tholdoffset": 7000
  },
  "ep": 1,
  "etag": "9b1f2fe8460510cca04b173c0ae4",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_motion.aq2",
  "name": "Motion Sensor",
  "state": {
    "dark": true,
    "daylight": false,
    "lastupdated": "2020-03-14T18:39:45",
    "lightlevel": 8451,
    "lux": 7
  },
  "swversion": "20170627",
  "type": "ZHALightLevel",
  "uniqueid": "00:15:8d:00:14:88:c2:aa-01-0400"
}
```

Can you check in the complete JSON ?
The lux value is not in the "config" field but the "state" one.

elpaso666, 2020-03-15:
I dont have anything else in Deconz-NodeRed in ZHALightLevel... it is also on github and I have been told there that the nodered deconz just taking what Deconz is sending

Smanar, 2020-03-16:
Take a look in the API, to checl if the problem is from deconz or nodered.
You haven't log or something else to find the url (or better the json directly) ?
The url to check is is something like (for all sensors)

`http://DECONZ_IP:DECONZ_PORT/api/API8KEY/sensors`

D3nnisd, 2020-03-16:
Which Sensor do you have? There are more of them and not all of them own a lux sensor...

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix