dresden-elektronik/deconz-rest-plugin #2588: Recovering manually from another filesystem

tuomassa, 2020-03-14:
My raspberry pi installation broke because of SD card failure probably, but I was able to copy the rootfs to safety. Now I installed new raspbian, and was wondering what folders I need to copy from old rootfs to new rootfs that all my zigbee stuff is automatically there and I dont have to reset and re-do everything?

ebaauw, 2020-03-14:
The network settings are stored on the device (RaspBee, ConBee), in non-volatile memory.
The REST API settings are stored in the deCONZ database.
After installing deCONZ, _before_ starting it, you want to restore everything under `~/.local/shared/dresden-elektronik/deCONZ`.

tuomassa, 2020-03-15:
Thanks so much! This worked

closed 2020-03-15 by tuomassa