dresden-elektronik/deconz-rest-plugin #2591: Aqara vibration sensor - After few hours it stop working (not send data)

SnaKKo, 2020-03-16:
I think the aqara vibration sensor have some issues with deconz; after a few hours my sensor is no longer sending data to deconz.
I tried to replace the battery, to rejoin the network, to set sensibility=1 but it work just for few hours.

Do you have any tip for me?

No issues with xiaomi door sensors, xiaomi temperature sensors, innr plug, ect....

ebaauw, 2020-03-16:
I have no issues with my `lumi.vibration.aq1`.

> Do you have any tip for me?

Not specific for this device, but in general, for Xioami sensors (that won't select a new parent):
- Pair the sensor close to its operational location (so it's parent remain within reach);
- Do not to power off any routers (lights) (so it's parent remains operational).

SnaKKo, 2020-03-16:
Thanks for the suggestions but I already follow these rules.
The device is reachable (In Phoscon App the sensor icon is not gray colored, but in Deconz GUI I don't see the line that connect the sensor to the coordinator) but the last-updated value is really old (also 5-6 hours) and also if I rotate/move/sollecitate the sensor it doesn't send data anymore.

I keep trying.

mickejo, 2020-03-20:
I have a similar issue reported here:
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2581](url)

Smanar, 2020-03-20:
@SnaKKo the name is not grayed in the GUI (even it's not in Phoscon) ?
Make a try with leting the device closer the conbee for some days.

SnaKKo, 2020-03-20:
> I have a similar issue reported here:
> [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2581](url)

> I have a similar issue reported here:
> [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2581](url)

FFY
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2581

SnaKKo, 2020-03-20:
> @SnaKKo the name is not grayed in the GUI (even it's not in Phoscon) ?
> Make a try with letting the device closer the conbee for some days.

I will make a try to bring near the device to the coordinator (ConBee II) thanks.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

sophof, 2020-11-10:
I suddenly have the same issue at the moment. It is just the aqara vibration sensors. They are all reporting status "Not reachable", but are connected fine. In home assistant the states are stuck to "off" and the attributes also don't seem to be updated. Adding the sensors again is no issue, but they still don't work afterwards.

Mimiix, 2020-11-10:
@sophof Please check https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3455 

Or open a new issue if not sufficient.