dresden-elektronik/deconz-rest-plugin #2592: IKEA Trådfri On/Off Plug switching power on/off/on/off when joining the mesh

rasssta, 2020-03-16:
Hi,

I have two Trådfri outlet On/Off Plugs connected to a Conbee II. Every time I connect them to power they switch the power on/off/on/off/etc like 5-10 times when joining the zigbee mesh. If I have the conbee GUI open I can see that they send some sort of signal every time they perform the on/off.

This is not a big issue when setting them up for the first time, even if it's not ideal. However, when you've had a power outage and they act like this when power is restored, the equipment connected to the plugs aren't very happy about it.

I'm running 2.05.74 with firmware 26530700 on Conbee and 2.0.022 on the Trådfri plug.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

rasssta, 2020-06-08:
I still see this behavior, even after upgrade of the outlets to 2.0.024

https://www.youtube.com/watch?v=M8EJc-Hdovg&feature=youtu.be

Mimiix, 2020-06-09:
Interesting behaviour, I'll check if others might have the same or not.

Mimiix, 2020-06-09:
I've spoken to others and they seem to have the same on this exact firmware. Can you post logs?

rasssta, 2020-06-09:
Sure. Which kind of logs do you want?

Mimiix, 2020-06-10:
deCONZ logs

rasssta, 2020-07-01:
I have not been at the location for a while. Will try to get the logs this weekend.

closed 2020-08-01 by stale[bot]