dresden-elektronik/deconz-rest-plugin #2594: Allow to specify "discovery address"

iwwwwwwi, 2020-03-16:
Hi,
First of all - thank you guys for the awesome piece of hardware+software. I absolutely love how it works and how nicely it integrates with Home Assistant. 
I'm using Phoscon running in a container using https://hub.docker.com/r/marthoc/deconz. It works with no issues. I recently switched to kubernetes, and also moved away from host network so the container gets kind of a random IP from a pool every time the pod is removed and recreated. This is causing HA to loose integration with phoscon since it has learnt the local IP of the container during the setup of integration. 
As Kubernetes provides this nicely working DNS resolution for all of the pods (that doesn't change during pod recreation) it would be nice to start deCONZ with a flag that allows to override the address that gets announced when "Authenticate app" is enabled.

Again - thank you, and keep up the good work ;)

Regards
Marcin

iwwwwwwi, 2020-06-29:
Hi there,
any idea if this ever gets any attention ? :) I'm just curious and don't want it to be automatically closed.

Mimiix, 2020-07-20:
Aah
Yes. I didn't get a notification for this. Sorry.

I'll add Backlog so the bot doesn't hurt you.

manup, 2020-07-21:
The easiest approach could be a command line switch. But it also opens up a security risk when the given address/hostname is accessible openly and not just by the HA instance.

Mimiix, 2020-07-21:
@manup This would be a risk that the user takes,as they shouldn't have ports opened.

Mimiix, 2020-07-21:
@iwwwwwwi You are able to edit the value in the zll.db file. I won't recommend this, but it is possible.

See table config2 in it.

iwwwwwwi, 2020-07-28:
@Mimiix i've updated ipaddress in the config2 table but although it doesn't change with restarts of deconz container i still get the dynamic IP when i configure the integration in home assistant. It appears it gets it from https://phoscon.de/discover so what i'd really like to do is to be able to manually specify content of what's getting set as "internalipaddress" in the discovery service.

Mimiix, 2020-07-29:
@iwwwwwwi I got the following reply from @manup 

10:38 AM] manup: The discover IP is the external IP that the discover server sees plus the internal IP which is reported from deCONZ (by going over the local network interfaces).
Currently it can't be specified, I'd like to add an command line switch for that.

I''ll add the Feature request label here :)

iwwwwwwi, 2020-07-29:
@Mimiix sounds good :) does it have to be an IP or a string with domain name would also work? (kubernetes serve their dynamic internal IPs via a static dns name)

markus7811, 2020-11-29:
Hi, i'm really interested in this solution. Deconz is a really great solution, but at the moment it's impossible for to run it in my microk8s cluster behind nginx ingress, because some "hard coded" ip adresses in hostnames won't make it work with dns. I think a proper solution would be to define the listen devices/ips per cli and remove all hardcoded ip addresses from the webinterface and replace it with relative links.

Xyaren, 2022-04-09:
I am too interested on this as I plan running the pwa via a nginx reverse proxy in my docker environment.

Please allow us to specify a public ip, port for discorvery or make it unnecessary for a locally deployed pwa via docker.
I dont't really care about http://phoscon.de/pwa:

The api at https://phoscon.my-domain.com/api/ actually responds, but the pwa is just not looking for that address, even if the pwa is served at https://phoscon.my-domain.com/pwa/login.html .
Instead it looks at http://10.181.255.2/api which is the wrong ip and protocol.

It would be nice to disable discovery and either allow us to configure api urls or even better deduct the api url from the pwa url.

Xyaren, 2022-04-09:
The hacks they established in order to change the pwa behaviour are amazing:
https://github.com/home-assistant/addons/blob/master/deconz/rootfs/etc/nginx/nginx.conf

manup, 2022-04-10:
There have been a bunch of changes so that the Phoscon App respects the URL it's called over for HA ingress and Reverse proxies. In recent versions you can also disable the discovery service via gateways advanced settings:

[image]

The discovery server URL is in the zll.db file (default https://phoscon.de/discover) you can change this to point to your own server but it needs to implement the response. For a pure local network with known IP/DNS names this isn't much useful.