dresden-elektronik/deconz-rest-plugin #2595: Log filled with "no button map for: SP 120 ep: 0x01"

jo-me, 2020-03-17:
Hi,
I'm running the latest version of Phoscon with latest version of Conbee
(2.05.75 and 26350500).

My log gets spammed with the following messages every second and I do not know why.
I have one Innr SP 120 added to the network, it is connected and always on.

Is this a bug?

```

15:08:29:211 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:29:211 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:30:219 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:30:219 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:31:227 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:31:228 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:32:236 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:32:236 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:33:212 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:33:212 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:39:214 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:39:215 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:40:223 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:40:223 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:44:224 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
15:08:44:224 ZCL attribute report 0x00158D0001FE558C for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
15:08:46:224 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
```

SwoopX, 2020-03-17:
Those can be ignored. Even though this is just a blemish, we might want to get rid of those messages by sharpening the corresponding code a bit.

@ebaauw @manup I guess this is the relevant line
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/171a1971f8bef1f7da0ff3fa54b8adb52d2ff90d/de_web_plugin.cpp#L743

Should we simply add a check for type "ZHASwitch" (not too sure if that would be sufficient or disregard some devices)?

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

meremortals70, 2020-10-04:
Still happening, the logs are full off them.

Smanar, 2020-10-04:
Don't worry, it s just a debug output.
Deconz don't use all return from device.

meremortals70, 2020-10-04:
Can it be turned off so the logs are readable?

Mimiix, 2020-10-04:
@meremortals70 please open a own user question issue.