dresden-elektronik/deconz-rest-plugin #2596: Question: Is it possible to create a scene w/o brightness?

jan666, 2020-03-18:
(Sorry if this is a duplicate or clearly impossible - but I can't find anything related to this)

I want to emulate the default ikea 5 button remote behaviour (changing the color temperature) via 5 pre defined scenes. So I created my scenes via:

```
curl -vv -XPOST .../api/<key>/groups/15/scenes -d '{ "name": "1" }'
curl -vv -XPOST .../api/<key>/groups/15/scenes -d '{ "name": "2" }'
```

and tried to just set the ct for the scene via

```
curl -vv -XPUT .../api/<key>/groups/15/scenes/1/lights/1/state -d '{ "on": true, "transitiontime": 10, "ct": 490 }' 
curl -vv -XPUT .../api/<key>/groups/15/scenes/2/lights/1/state -d '{ "on": true, "transitiontime": 10, "ct": 380 }' 
```

but the resulting scene is

```
{
  "lights": [
    {
      "bri": 249,
      "colormode": "xy",
      "id": "1",
      "on": true,
      "transitiontime": 10,
      "x": 0,
      "y": 0
    }
  ],
  "name": "1",
  "state": 0
}
```

with brighness 249. Is it even possible to just change the temperature via scene and leave the brigness in the state it was?

Mimiix, 2020-06-06:
@jan666 is this already solved?

jan666, 2020-06-06:
No, looks like no one had an idea

Mimiix, 2020-06-06:
Ah. Could the BRI be out of bounds for the device? The api will return a MIN max CT for the device.

Did you check that?

closed 2020-07-11 by stale[bot]