dresden-elektronik/deconz-rest-plugin #2597: IKEA Blind not showing up in the deconz page, but does show up on the graphical representation

fribse, 2020-03-18:
I am running deCONZ as a HomeAssistant addon.
I have tried adding an IKEA repeater, IKEA blind and IKEA controller to Deconz.
When I view the deconz connections in the graphical via VNC then all the items show up.
But if I view via the Deconz webinterface I don't see the blind, my guess is that it would appear under lights.

I can't get the blind to react either, so my wish was to remove it from the zigbee network, and then try adding it again, but I don't see how to do that, when it doesn't show up in the first place.

The VNC view:
[image]

fribse, 2020-03-19:
I have tried re-adding the 'range extender', the controller and the blind. I can get the controller paired with the range extender, and the blind, and it does show up in the VNC console, but I can't make it react through the deconz, it works through the IKEA remote control.

fribse, 2020-03-19:
Probably the same as #2558, mine is also 2.05.75

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix