dresden-elektronik/deconz-rest-plugin #2598: Question : pairing procedure

guillaume1388, 2020-03-18:
Hello,

I am trying to use your rest api to manage zigbee devices. I have a question about pairing procedure.

What are the commands to send to the api ?

I have no problem to pair devices with phoscon app. But I would like to pair devices whitout phoscon, by using postman for example.

On phoscon app, I have to click on "add new sensor" button, I select the manufacturer and 3 minutes are availables to put the device in pairing mode. I would like to know what are the commands sent during these actions in order to pair the device whitout phoson app.

Best regards,

Guillaume Majorel

Smanar, 2020-03-18:
It's the "permitjoin" field in "config"

permitjoin | Number (0..255) | Open the network so that other zigbee devices can join. 0  = network closed, 255 = network open, 1..254 = time in seconds the  network remains open. The value will decrement automatically.
-- | -- | --

guillaume1388, 2020-03-19:
thank you ! That works !

closed 2020-03-19 by guillaume1388