dresden-elektronik/deconz-rest-plugin #2601: New Device Request - Sonoff S31 Lite zb

Bitmaster20, 2020-03-19:
Product page can be found here
https://sonoff.tech/product/zigbee-smart-plugs/sonoff-s31-lite-zb

[image]
[image]
[image]

Bitmaster20, 2020-03-19:
Devices seem to show up in Phoscon so I'm not sure if there is any further coding needed.   However its not listed on the supported devices page.
[image]

SwoopX, 2020-03-19:
It doesn't have any sensors, so that's beneficial for you. On/off functionality is almost always supported out of the box. Since the device seems to be working, there's nothing left to do.

Bitmaster20, 2020-03-19:
Thanks.  I’ve tested this and it’s triggering on/off without issues.  Am I allowed to edit the supported devices page to add as a supported device (which I don’t know if I have permission or access to do) or is there a formal process to go through first?

SwoopX, 2020-03-19:
Well, afaik, it is generally free to edit.

Bitmaster20, 2020-03-20:
I've added it as a supported device

closed 2020-03-23 by Bitmaster20