dresden-elektronik/deconz-rest-plugin #2602: "on" is ignored if send together with "bri"

lukicsl, 2020-03-19:
**1.  the lights of the group are ON**

When I  send a json to a group in form:
`{"on":false,"bri":30}`
>the lights of the group stay ON value changed to 30
EXPECTATION: Light should should go OFF

**2. the lights of the group are OFF**

When I  send a json to a group in form:
`{"on":false,"bri":30}`
>the lights of the group go ON with value 30
EXPECTATION: Should not go on

BUT WHEN:

**3. the lights of the group are OFF**

When I  send a json to a group in form:
`{"bri":255}`
>the lights of the group stay off
EXPECTATION: should stay off

THEN
When I  send a json to a group in form:
`{"on":true}`
>the lights of the group go ON with value 30
EXPECTATION: Should go on with value 255

Are my expectations ok?
If this is true then we have an issue.
I did not test against a true hue bridge.

ebaauw, 2020-03-19:
The REST API plugin handles these requests differently from the Hue bridge and, in my opinion, incorrectly.  I’ve been wanting to correct this for a long time, but it’s quite a significant change.  See also #2553.

1.  Lights should dim to 30 and then turn off.  With next on, they should go on at 30.
2. Lights should stay off.  The bri is probably ignored by the lights.  When sent to a single light, error is returned that you cannot change bri while the light is off.  Not sure if similar error should be given for group commands (based on all_on).
3. Lights should stay off.  Bri is probably ignored by the lights.
4. Lights turn on at previous level.  Might depend on how light deal with receiving bri while off.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix