dresden-elektronik/deconz-rest-plugin #2604: Philips Hue E27 Brightness issue when switching ON

mike9011, 2020-03-19:
## Information
Product: ConBee II
Version: 2.05.75 / 3/8/2020
Firmware: 264A0700

#### Philips Hue Bulb: 
Model: LWW001 (E27)
Version: 1.50.2_r30933

## Issue
When i turn on the light with brightness 80, it dims up > dims down to [x] > dims up to 80.
This issue occurs also when i don't set any brightness, only turn the light on.
When i send a command from the Phoscon GUI it gives the same issue.

## Video
https://photos.google.com/share/AF1QipOBbZFTjCpfr8cAkd4o1GoYEkyeLdY56I1M7I_1zilafas7kz7fXMYFOsQ017NVdQ?key=Nm5QUXotcHcyaURXcHFydXBsZ051M3lYTkJSaDBB

## Screenshots
[image]
[image]
[image]

mike9011, 2020-03-19:
Issue seems to be resolved after resetting the bulb to factory settings.

mike9011, 2020-03-19:
Additional info:

After a factory reset of the bulb it worked perfectly fine. Unfortunately after a couple hour and a few tests later the problem is back.

Mimiix, 2020-06-05:
@mike9011 did you figure it out?

mike9011, 2020-06-05:
@Mimiix unfortunately not.. I replaced the bulb with a colored hue. Still the same issue with the white light version.

mike9011, 2020-06-26:
Issue is not resolved

Mimiix, 2020-07-20:
@SwoopX Can you have a look into this if you got some spare time :)?

mike9011, 2020-08-13:
Can someone remove this bot from this issue?

Mimiix, 2020-08-14:
@mike9011 I can:) But it also keep people active (like me now getting the notification). I can make it "Backlog".

Xylomid, 2021-01-17:
did you find a solution ? I have the same problem with E27 Hue. Thanks

mike9011, 2021-01-17:
Unfortunately not, its still on their to-do list after a year ;-)

ebaauw, 2021-01-17:
Can you log what API commands you send to turn on the light?  And to change the brightness?  Are you still on v2.05.75?

manup, 2025-05-02:
Clsoing, the Hue DDFs now also provide configuration for power on state.

closed 2025-05-02 by manup