dresden-elektronik/deconz-rest-plugin #2605: Renaming and deleting issue

SAOPP, 2020-03-19:
Hi!

Guys, I'm faced with the strange behaviour of my Raspbee running in addon of HA. When I rename some device, and after this click on reboot addon or for example shutdown the instance, after boot I see the old name of my device, why this happens?

And another issue with deleting the devices. Few weeks ago I moved three devices from my raspbee to conbee2, and again, after my HA instance is reboot (or addon/container with raspbee was rebooted), I see in phoscon my three old devices that I deleted it before as unreacheble. What's I do wrong?

Also, I see two of them, at my raspbee zigbee map. Not understand whats goiong on? What I need to do for property delete this devices and what I need to do with renaming of my first question? Thanks in advance!

SAOPP, 2020-03-20:
Guys, any comments? :| Maybe needs some additional info or what? 'Coz the first issue with renaming is looks like not a big, but the second issue I hope I can to solve it 'coz this is unreal headache for me. :/

Smanar, 2020-03-20:
Where are you renaming the device ? The name have changed on the API ?

How are you deleting your device, are they visible in the gui, grayed or not ?

SAOPP, 2020-03-20:
Hi! Thanks for reply!

I also started a few days ago another topic, and there is information related to my current question, we can say these topics should be connected.

About your question how I delete my devices... I do it in Phoscon, first I'm delete the device from my Rapsbee, and also restart the deconz addon. After this I'm paired device to my Conbee2.

About API, u mean do i need to see in rest api for current device names? They had changed names or not? I haven't done this before, but I can do this right now with one of the sensors, which I need to rename. And post the result.

Just strange behaviour, 'zoc damn, I deleted the device, why raspbee is not save this changes? :( In Okay, I'll go to rename the device and look at the rest-api...

Smanar, 2020-03-20:
If you are using phoscon and not HA, the problem is from deconz.

But there is something I m not sure, you have 2 differents machines or you have the conbee and the raspbee on the same ? and you just swap them ?

Smanar, 2020-03-20:
Ok so I have tried on my side

- Use phoscon to chnage a sensor name
- Close deconz, and reload it (IDK what you are calling the "deconz addon")
- Open Phoscon again, the name is still the last.

BTW , you haven't aswer, do you use 2 machines ?

SAOPP, 2020-03-20:
> If you are using phoscon and not HA, the problem is from deconz.
> 
> But there is something I m not sure, you have 2 differents machines or you have the conbee and the raspbee on the same ? and you just swap them ?
> 
> You are renaming your device in phoscon ? And if you restart deconz, the old name is back ?

I'm using the HA, my Raspbee is working like a HA addon on the same Instance. And my Conbee2 working on another instance in docker and connected with my HA via integration...

SAOPP, 2020-03-20:
deconz addon - I mean addon for HA (before it called HASSIO).

Smanar, 2020-03-20:
So to resume you have on the same machine
- 1 docker with HA + 1 deconz + Raspbee
- 1 docker with 1 deconz + 1 conbee.

SAOPP, 2020-03-20:
[image]

SAOPP, 2020-03-20:
> So to resume you have on the same machine
> 
> * 1 docker with HA + 1 deconz + Raspbee
> * 1 docker with 1 deconz + 1 conbee.
> 
> If you relaod the HA addon it change nothing on deconz API, adn (ofc deconz database)

Yes this scheme is right.

SAOPP, 2020-03-20:
Rpi3B with Raspbee and HA.
Another x86 mini pc with docker and conbee2.

Smanar, 2020-03-20:
What is the 00212EFFF0..... it's the mac adress of the device ?
They are different ?

Smanar, 2020-03-20:
> 
> 
> Rpi3B with Raspbee and HA my main HA instance.
> Another x86 mini pc with Docker and Conbee2 - just my mini server...

Ok so, no, you are using 2 differents machines.

SAOPP, 2020-03-20:
> What is the 00212EFFF0..... it's the mac adress of the device ?
> They are different ?

Different of course. I don't do any restore of backups from one to another.

SAOPP, 2020-03-20:
just started using my conbee 2, and start to move some sensors near conbee 2 to this coordinator :)

Smanar, 2020-03-20:
So I don't see something bad on your setup, sorry.

You can try the same test than me (with realoading only deconz) but your name bug is strange.

For the "ghost" sensor, the name isn't grayed now (since the time you have removed it) on the GUI ?
If yes just select it and press delete.

SAOPP, 2020-03-20:
...and faced with issue with deleting, for example if power outage is gone, and my main instance with rapsbee is rebooting, I have a trouble with old "dead" not reachable devices in my raspbee.

SAOPP, 2020-03-20:
> So I don't see something bad on your setup, sorry.
> 
> You can try the same test than me (with realoading only deconz) but your name bug is strange.
> 
> For the "ghost" sensor, the name isn't grayed now (since the time you have removed it) on the GUI ?
> If yes just select it and press delete.

Yes, ghost devices back after full restart my instance. And I need to delete it from raspbee phoscon, and restart HA again, after this my instance can working with this devices and corrent reaching it from conbee2, coz I have the same names in conbee2 phoscon as they have it on my raspbee before. Understood the issue?

Smanar, 2020-03-20:
I think yes, It's a kind of database problem (and you have it for name and complete device config), but I realy don't have idea and can't reproduce the bug on my side.

Have you tried to explain your bug on deconz docker github ?

I can't say you if it's from deconz or docker setup, but I think it's better exploring the 2 ways. And ATM you are the only one with theses stranges bugs here.

Smanar, 2020-03-20:
BTW have you tried to deleting your device, with deleting them from deconz UI  (and phoscon after if needed) ?

SAOPP, 2020-03-20:
Not sure that I'm understood our question correct. U mean delete device from phoscon or what? Or maybe old phoscon gui or? I delete all my devices via phoscon web gui. Or u mean vnc? Please explaine. :)

SAOPP, 2020-03-20:
About ask in docker deconz, yes, I think need to ask in addon git repo... Thanks.

SAOPP, 2020-03-20:
Also the channel is difference, raspbee is working on 11, and conbee2 on 15th channel. It's just for additional infos about my config.

SAOPP, 2020-03-20:
> I think yes, It's a kind of database problem (and you have it for name and complete device config), but I realy don't have idea and can't reproduce the bug on my side.
> 
Actually yep, I'm thinking like this too, db issue, something like permission denied for writing, not sure but seems like that. These are just my thoughts in the air....

SAOPP, 2020-03-20:
I'm already starting to think about resetting raspbee, and moving the board to a stand-alone instance. but damn, I would not really like to bother with pairing devices - although of course I can try to deploy a backup on a new device. Well, summing up a certain result, I continue to think that software is to blame for everything, and the latest versions of the phoscon or container ... somewhere they broke something. Well, or this is really my local bug.

Smanar, 2020-03-21:
> 
> 
> Not sure that I'm understood our question correct. U mean delete device from phoscon or what? Or maybe old phoscon gui or? I delete all my devices via phoscon web gui. Or u mean vnc? Please explaine. :)

Yep, using the vnc, to delete device in deconz application (with pressing delete) but it can not work, and better to wait for the device have his name grayed (unreachable).

>Actually yep, I'm thinking like this too, db issue, something like permission denied for writing, not sure but seems like that. These are just my thoughts in the air....

It's for that I think it s a good idea to show your bug on docker github, they know better their installation, but even with docker, it the first time I m seing this issue.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix