dresden-elektronik/deconz-rest-plugin #2606: New Device Request for Hive Motion Sensor

eddiehowrya, 2020-03-19:
This request is for the Hive motion sensors which are part of the British Gas family of smart devices.
https://www.hivehome.com/products/hive-motion-sensor?cid=ppc.goo.bran....&icid=mname%3Amega-menu.iname%3Ahive-motion-sensor

Node Info
[image]

Occupancy Cluster
[image]

Power Config Cluster
[image]

Temperature Cluster
[image]

Basic Cluster
[image]

Illuminance Cluster
[image]

SwoopX, 2020-03-19:
I'll put it on my list.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

eddiehowrya, 2020-06-07:
Sensors are working well with the new version, thank you.

lwsrbrts, 2021-02-06:
> 
> 
> Sensors are working well with the new version, thank you.

@eddiehowrya - apologies for responding on a closed issue. I wondered if you had done anything specific with your Hive sensors after pairing them? I have a sensor that seems to be self-activating - I only have the one Hive motion sensor so don't know if this affects others.

It detects motion for one minute which then clears for 4 minutes and then it detects motion again for a minute...so on and so forth.

Screenshot from Home Assistant of it activating itself overnight.
[image]