dresden-elektronik/deconz-rest-plugin #2607: (Q) on the fly replace ConbeeII with RaspBeeII?

salopette, 2020-03-19:
Hello, I got a RaspBeeII, can I just replace it instead of ConBeeII?
Do all devices and nodes remain?

Smanar, 2020-03-19:
I think almost yes.

But make a backup first, there is some data on the raspberry but there is some data on the device too (network setting), so it can work, but better make backup/Restore.

salopette, 2020-03-19:
Ok, thx! I check it tomorrow!

Smanar, 2020-03-19:
And seriously don't forget the backup, realy important.

Smanar, 2020-03-19:
And, I m changing my answer ^^.

No, It will not work.

I m looking the new device, there is so much difference in installation https://phoscon.de/en/raspbee2/install

ebaauw, 2020-03-19:
You can replace a broken coordinator device in a network (although I never tried it).  The ConBee II would take over the identity (mac address, extended PAN ID, etc) of the RaspBee, and the other devices in the network won’t see any difference.

However, you can no longer use the RaspBee within range of the network, as this would cause an address conflict.  If you want to be able to use both devices, you need the ConBee II to setup a new network and reset all devices and pair them with the new network.

salopette, 2020-03-19:
Even if I want to use conbee with Zigbee2mqtt nearby?

salopette, 2020-03-19:
> Übrigens, wo hast du eine Raspbee 2 gesehen?

https://www.amazon.de/gp/product/B084MK8F5M/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

manup, 2020-03-19:
And this product page for more details: https://phoscon.de/raspbee2

Currently I wouldn't recommend the backup from ConBee II restore to RaspBee II approach since we still figuring out to get this work smoothly.

Smanar, 2020-03-19:
> Currently I wouldn't recommend the backup from ConBee II restore to RaspBee II approach since we still figuring out to get this work smoothly.

@manup So the backup (with conbee 2) and restore (on Raspbee) can be a bad procedure ?

ebaauw, 2020-03-19:
> And this product page for more details: https://phoscon.de/raspbee2

Cool!  It looks shorter than the RaspBee, might have less of an impact on the RaspBerry's thermal management.  Is the real-time clock exposed over the serial protocol as well, or does it have a separate interface?  Does it have the same pin layout as the RaspBee?  Does it still use GPIO 17 for reset?  Does it have the same radio hardware (and firmware) as the ConBee II?

Smanar, 2020-03-19:
Ha yes, good question, the firmware ?

manup, 2020-03-19:
> Cool! It looks shorter than the RaspBee, might have less of an impact on the RaspBerry's thermal management. Is the real-time clock exposed over the serial protocol as well, or does it have a [...]

Yes, the dimensions were chosen to that it better fits on a Raspberry Pi Zero. We haven't testet it for the thermal differences compared to the RaspBee I yet, but that's a good idea I'll forward that to the team.

>Is the real-time clock exposed over the serial protocol as well, or does it have a separate interface?

The RTC is managed directly the standard Linux kernel driver and communicated via I2C. The Radio-MCU itself doesn't have access to the RTC.

Note the kernel driver needs to be compiled and activated manually with help of this repository:

https://github.com/dresden-elektronik/raspbee2-rtc

>Does it have the same pin layout as the RaspBee? Does it still use GPIO 17 for reset?

Besides the I2C pins the Rx/Tx and Reset pins are identical.

>Does it have the same radio hardware (and firmware) as the ConBee II?

Kind of :) same MCU as the ConBee II and the goal is to just have one firmware but we're still figuring out a few issues. The firmware version 0x26530700 for example supports both ConBee II and RaspBee II, but seems to have issues when updated from former ConBee II firmware.

One notable thing is the new bootloader which is highly improved to be reliable on startup. It will later be integrated in ConBee II as well and we're making tests to upgrade existing bootloaders of ConBee II (which is fairly tricky to get right without leaving the chance to bricking a device).

In the coming days we'd like to send free sample to project maintainers :)

manup, 2020-03-19:
> > Currently I wouldn't recommend the backup from ConBee II restore to RaspBee II approach since we still figuring out to get this work smoothly.
> 
> @manup So the backup (with conbee 2) and restore (on Raspbee 2) is not advised ?

It can be tried and for some it worked pretty good but some tests did't pick up the network for 30 minutes or not at all. So currently this is very experimental.

Smanar, 2020-03-19:
Arf, bad news for for those in a hurry to change.

BTW, I think the smaller size to be used on a pi 0 is a realy good idea.

SwoopX, 2020-03-19:
@manup would the new hardware also include some range benefits? Or maybe some performance improvements? Asking because of my production Pi Zero ;)

salopette, 2020-03-20:
> Sie können RaspBee jedoch nicht mehr in Reichweite des Netzwerks verwenden, da dies zu einem Adresskonflikt führen würde.

Will there be address problems if I use it on different systems?

manup, 2020-03-20:
> @manup would the new hardware also include some range benefits? Or maybe some performance improvements? Asking because of my production Pi Zero ;)

In regards to range and Zigbee the RaspBee II is identical to ConBee II since it uses the same radio MCU and power-amplifier.

Compared to the RaspBee I 8-bit AVR MCU the RaspBee II MCU is faster, it's a 32-bit ARM core.
Although this won't be noticeable much since even the 8-bit AVR is fast enough to handle the Zigbee task.

manup, 2020-03-20:
> Will there be address problems if I use it on different systems?

Address problems only occur when two devices with the same MAC address are operating in close range. You can check the MAC address in the deCONZ network settings.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix