dresden-elektronik/deconz-rest-plugin #2609: apt repo down

thorus90, 2020-03-20:
Currently http://phoscon.de/apt/deconz shows 404 , but accoring to https://phoscon.de/de/conbee2/install#raspbian this is still the correct URL.

ebaauw, 2020-03-20:
Same here, stretch as well as buster; main as well as beta repositories.

manup, 2020-03-20:
Sorry, the files didn't completely update after the transfer to our new server a few hours ago.
It should work again now.

thorus90, 2020-03-20:
Thanks works again

closed 2020-03-20 by thorus90

martindederer, 2020-06-07:
@manup Looks like http://phoscon.de/apt/deconz returns a http status code 403 now.

manup, 2020-06-07:
Just tried it `apt install deconz` works here. Or do you mean access via browser?

martindederer, 2020-06-07:
I was getting a 403 when i tried to access the directory through a browser. Also apt complains about a missing "Release" file. I am not sure though. Is this the expected behavior?

```
$ sudo apt update
Ign:1 http://phoscon.de/apt/deconz eoan InRelease
Err:2 http://phoscon.de/apt/deconz eoan Release                                                    
  404  Not Found [IP: 144.76.96.194 80]
...
Reading package lists... Done
E: The repository 'http://phoscon.de/apt/deconz eoan Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

The deconz package is not available after executing the apt update command. I just followed the installation instructions on https://phoscon.de/en/conbee2/install#ubuntu. Did i miss something?

manup, 2020-06-07:
I think it's the Ubuntu version 19.10 isn't in the APT repository.
However since its bionic is upward compatible, can you please update the `/etc/apt/sources.list.d/deconz.list` file:

```
deb [arch=amd64] http://phoscon.de/apt/deconz   bionic main
```

or for the beta version:

```
deb [arch=amd64] http://phoscon.de/apt/deconz   bionic-beta main
```

martindederer, 2020-06-07:
Nice, that solved the issue. Thank you for your prompt assistance. :+1:

psoares, 2021-04-19:
I'm trying on Ubuntu groovy on a raspberry pi 4 and get 2 issues:
1/ Repo is insecure:
```
E: The repository 'http://phoscon.de/apt/deconz groovy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.```

2/ I updated `/etc/apt/sources.list.d/deconz.list` to be like this:
```
deb [arch=aarch64] http://phoscon.de/apt/deconz             groovy main
```
Then issued:
```
sudo apt update --allow-unauthenticated
```
The result was:
```
Err:7 http://phoscon.de/apt/deconz groovy Release
  404  Not Found [IP: 144.76.96.194 80]
```

100hp4ever, 2021-11-16:
http://phoscon.de/apt/deconz returns 403 forbidden

xsirpreisx, 2021-11-27:
E: Failed to fetch http://phoscon.de/apt/deconz/pool/main/d/deconz/deconz_2.12.06-raspbian-buster-stable_armhf.deb  Could not connect to phoscon.de:80 (144.76.96.194), connection timed out

Has this URL changed? The whole phoscon.de website seems to be offline?

Smanar, 2021-11-27:
>The whole phoscon.de website seems to be offline?

Yep, it's exaclty that, need to wait for return.

xsirpreisx, 2021-11-27:
It's working now. Just bad timing from my part, I guess ;-)

fritolays, 2021-12-07:
Once again down...