dresden-elektronik/deconz-rest-plugin #2610: dtoverlay option gets overwritten on each update

Hypfer, 2020-03-20:
Every time I update deconz using the package manager, other software on the pi breaks due to the replacement of `dtoverlay=pi3-disable-bt` with `dtoverlay=pi3-miniuart-bt-overlay`.

Please stop doing that.

manup, 2020-03-20:
The reasoning for that is to be able to allow Bluetooth and RaspBee (UART) to be used at the same time. We can disable this automatic overwrite and document it to be configured manually (need do do some more tests first).

Out of interest, it shouldn't break stuff: Which applications are affected in your setup?

Hypfer, 2020-03-20:
Maybe you could use a `dtoverlay` whitelist and only overwrite if the value disables UART?

The reason is quite frustrating actually:
Since bluetooth adapters do tend to break randomly requiring a hard power cycle, I'm using two usb bluetooth adapters in combination with the software-switchable usb hub of the pi3b+

If the internal bluetooth is enabled, `hci0/1/2` get shuffled around on each boot for some reason. Probably a race condition of some sorts.
Sadly, there seems to be no way to get predictable bluetooth interface names like we have with network interfaces, which I need, because devices are paired with a specific adapter.

It might probably be possible to fetch all hci adapters and get the current `hcix` name by mac, but it's easier to simply disable the internal bluetooth since the usb bluetooth adapters do apparently always enumerate in the same order.

It's this thing btw: https://github.com/Hypfer/Cybele

Hypfer, 2020-04-17:
It happend again.

The problem is, that the pairing data of a specific peripheral (glance clock) seems to get corrupted somewhere somehow when that happens.
I think it's a failsafe in its firmware that simply deletes stored authentication keys or something when requests are received from a different bluetooth device than expected.

Pairing that thing is a pain every time. Please stop breaking it :(

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

Hypfer, 2020-06-06:
Wtf.
Stop closing issues without even checking their age and/or relevance.

Yes this is still an issue but I don't have the ability to reopen it

Hypfer, 2020-06-27:
ffs

Hypfer, 2020-07-26:
ffs