dresden-elektronik/deconz-rest-plugin #2611: Upgrading Osram / Ledvance smart plugs with image 0x0000

Schaussi, 2020-03-21:
Hi,

in my environment (Conbee II, RPi4, official current Raspbian image), devices like Xiaomi motion sensors sometimes get "lost" (I have to pair it again), or I get the sensor signals later (or the device does not react for e.g. 30 sec).
So I guessed it would be a good approach to update all wired Zigbee plugs and lights.
Unfortunately, I have not been able to start device firmware updates yet.
Regarding Osram plugs, I followed the official deconz upgrade procedure, but deconz GUI shows only an image type "0x0000".
Manufacturer code in Node info is 0xbbaa, OTAU Update tab shows 0x1135. 

Also trying upgrading other devices, the Query button seems to have no effect.

Please give me some hints what to do to receive the upgrades.

Thank you!

BastianWie, 2020-03-21:
@Schaussi it seamed there are two different Osram devices in the marked. The ones officially produced by Osram (firmware can be found [here](http://s3-eu-west-1.amazonaws.com/lightify-osram-firmware/)) and the one produced by Ledvance in the name from Osram. Only the last ones can be updated currently with following the Phoscon howto [here](https://phoscon.de/en/support#ota-update-osram-devices).

However I feel as well that there is a bug somewhere as the version is outlined as 0x00000 as well as the image... which seamed to be exactly what you see here.

By the way I´m using the Phoscon RaspBeeII with Raspbian Buster Desktop (stable 9-2019) installed via this howto [here](https://www.admin-enclave.com/en/articles/linux/493-how-to-connect-a-osram-on-off-plug-with-phoscon.html).

[image]

ebaauw, 2020-03-21:
> However I feel as well that there is a bug somewhere in deConz as the version is outlined as 0x00000 as well as the image

That's not a bug; it simply means that the device hasn't yet issued a request to check for new firmware.  Normally, pressing _Query_ while the device is selected in the _STD OTAU plugin_ triggers this, but not always (e.g. when network is open, or when device is unreachable).  Sometimes power cycling the device also triggers this.

> Keep also notes that there are "Osram Lightify" and "Osram Smart+" devices out there:

The older Lightify plugs have image 0x000a; the newer Smart+ plugs have image 0x0027.  So it's a bigger change than just rebranding.  Both use manufacturer code 0x110C for the image;  the _Node Descriptor_ advertises manufacturer code 0xBBAA.

I didn't know the Lightify firmware site on Amazon, but some of the firmware there also seems to be published to the Ledvance firmware site.  The `_encrypt.bin` files from the Lightify site don't appear to be valid OTA files, or they are indeed encrypted, incl. the OTA header.

BastianWie, 2020-03-21:
> That's not a bug; it simply means that the device hasn't yet issued a request to check for new firmware. Normally, pressing _Query_ while the device is selected in the _STD OTAU plugin_ triggers [...]

Oh you are right (removed the plug from the wall and put it back after 4min) now it shows:
- Version: 0x01320490
- Image: 0x0027

> I didn't know the Lightify firmware site on Amazon, but some of the firmware there also seems to be published to the Ledvance firmware site. The `_encrypt.bin` files from the Lightify site don't appear to be valid OTA files, or they are indeed encrypted, incl. the OTA header.

Its the official Osram site (sniffed from there HTTP traffic) which is used by the Osram Gateway Plugin. As [Osram no longer support there ecoystem it will be shutdown in late 2021](https://www.osram.com/cb/lightify/index.jsp) I assume.

Schaussi, 2020-03-21:
I could do the upgrades now (see my comments in first post).
As I remember, all devices having connections were connected to the plugs which showed up unversioned (Ledvance Smart+, 0x0000).
So by applying the 0x0027 on them, I think there is a good chance the problem will be gone.
Thanks guys for leading me in the right direction.

closed 2020-03-21 by Schaussi

BastianWie, 2020-03-21:
You where able to flash the Osram Smart+ devices as well? Which image did you used from the Ledvance website?

ebaauw, 2020-03-21:
> Manufacturer ID: 0x1135

That's the (hard-coded) default Manufacturer ID for the file, not for the device.  To wit: when you select a different node, it still displays this code.  The plug uses 0x110C.  I'll update the STD OTAU plugin to display the device code in the table.

> You where able to flash the Osram Smart+ devices as well? Which image did you used from the Ledvance website?

Months ago already.  See above: image 0x0027.  Search for the image type on the Ledvance portal.

BastianWie, 2020-03-21:
>  I'll update the STD OTAU plugin to display the device code in the table.

thanks 👍 

> Months ago already. See above: image 0x0027. Search for the image type on the Ledvance portal.

I found my issue. The device still showed the version V1.04.90 in the GUI while it was already updated from 0x01020490 to 0x01020509, which got me somehow confused. I performed a reset and re-paired the device now it shows a V1.05.09 version in the GUI. So everything is fine... I didn´t assumed that it required a reset + repair to show the new version in the GUI.

ebaauw, 2020-03-21:
See https://github.com/dresden-elektronik/deconz-ota-plugin/pull/3.

BastianWie, 2020-03-22:
@ebaauw  thanks :)