dresden-elektronik/deconz-rest-plugin #2614: Suddenly firmware not connected

daniel-lo, 2020-03-21:
My working installation had suddenly died .. not aware of any changes .. what can I do about this?
[image]

Smanar, 2020-03-21:
Have you another USB port to test ?
What is your OS ?
Can you try this command 
`GCFFlasher_internal -l`

daniel-lo, 2020-03-22:
It healed itself over night .. I really have no idea what happened.
How doe I ussue that command .. from Ubutu shell?

I wonder why it tells me firmware 2.5.74 is up to date while "Little Walter ⋅ v2.5.75" is listed in the change history?!

Smanar, 2020-03-22:
2.5.74 is the deconz version, not the firmware.
You can update deconz if you need it.

Yep this command is for use in a shell.
It can say if the system detect the device or not, it's just an informative command.

daniel-lo, 2020-03-22:
Next I would have rebooted the machine .. but since it has recovered I see no reason.
How do I update DeConz .. I did an ubuntu update/upgrade yesterday but it still shows .74.

Smanar, 2020-03-22:
Perhaps it's because stable/beta channel ?
You have all files here http://deconz.dresden-elektronik.de/ubuntu/beta/

daniel-lo, 2020-03-22:
could you point me to update instructions? I should be on stable .. "Little Walter" sounded stable.

Smanar, 2020-03-22:
For exemple

```
$> wget http://deconz.dresden-elektronik.de/ubuntu/beta/deconz-latest-beta.deb
$> sudo dpkg -i deconz-latest-beta.deb
$> sudo apt update
$> sudo apt install -f
```

daniel-lo, 2020-03-22:
thanks .. did so with deconz-dev-2.05.75.deb. Process ran well. But I still see
2.05.74 in my phoscon app.

Smanar, 2020-03-22:
Ha ?
So it's perhaps normal ^^, it's perhaps for that I m seing 74 too, small bug.

daniel-lo, 2020-03-22:
do .. I simply followed your steps .. then rebooted

Smanar, 2020-03-22:
ha ok, so not sure the OS can overwrite the files when the app is already running.

daniel-lo, 2020-03-22:
pls add how to stop deconz

Smanar, 2020-03-22:
You have headless or GUI version ?

daniel-lo, 2020-03-22:
thanks .. tried it .. version number doesn't change .. maybe the string was not maintained.

Smanar, 2020-03-22:
After so many tries, I think yes ^^.

SwoopX, 2020-03-22:
@daniel-lo If I read correctly, you installed the development package, not deconz.

daniel-lo, 2020-03-23:
@SwoopX what makes you think that? At least not intentionally .. I simply wanted a straight forward stable installatino (gui version).

SwoopX, 2020-03-23:
> thanks .. did so with deconz-**dev**-2.05.75.deb.

Smanar, 2020-03-23:
Lol, ha yep, good eyes ^^

daniel-lo, 2020-03-23:
I see .. there is a dev and a qt file .. didn't notice that .. just went for the number. Will switch. Thanks

daniel-lo, 2020-03-23:
Now it shows the correct number. My bad (using the dev build) .. THANKS Y'ALL!

closed 2020-03-23 by daniel-lo