dresden-elektronik/deconz-rest-plugin #2616: Websocket seem to aggregate some events

idittansikte, 2020-03-22:
I am trying to implement a client that reacts on sensor values at real time with user interaction, specifically switch button presses. I have two types of switches, TRÅDFRI Wireless Dimmer and TRÅDFRI ON/OFF Switch. Have same problem with both.

Mostly, the events are propagated though websocket in less than a seconds from button press. However, quite frequently it takes about a minute to get the event(s) from websocket API after a button is pressed. When that delay happens, all switch button events come at once.

When this delay (i.e. large latency) happens, there are other events like temperature sensor values coming which is an indication that the connection between deconz and my client is fine and the behavior originates from deconz. I am tracing the events received directly after socket read and its there I can see this issue.

Seem like some events are aggregated/buffered for some events in deconz/ConBee. Is that so? This is a known problem or by design?

Deconz: 2.05.75 / 3/8/2020
ConBee I FW: 26350500

SwoopX, 2020-03-22:
Are the events (e.g. termperature) from the same device?

idittansikte, 2020-03-22:
> Are the events (e.g. temperature) from the same device?

No, I have two aquara temp sensors and the two ikea switches I mentioned before.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix